sinara-hw/Pounder #1: [RFC] New EEM: Pounder (PDH lock signal generator)

dtcallcock, 2019-06-01:
[Pound Drever Hall](https://en.wikipedia.org/wiki/Pound–Drever–Hall_technique) locks are pretty ubiquitous in laser labs.  It would be nice to have a Sinara board that does all the necessary modulation and demodulation.  

It would sit alongside one of the proposed servo cards such as [Stabilizer (Microcontroller Servo)](https://github.com/sinara-hw/meta/issues/27) or [Fast Kasli Lock](
https://github.com/sinara-hw/meta/issues/26).

There are commercial models available from [Toptica](https://www.toptica.com/products/tunable-diode-lasers/laser-locking-electronics/pdd-110f-pound-drever-hall/), [Also Toptica](https://www.toptica.com/products/tunable-diode-lasers/laser-locking-electronics/pdhdlc-pro-pound-drever-hall/), [Sacher](https://www.sacher-laser.com/home/lab-equipment/locking_equipment/servo_controller/top_of_line_modulator_demodulator-lb20.html), [Qubig](http://www.qubig.com/rf-drivers.htm), and [Stable Laser Systems](http://stablelasers.com/products/pound-drever-hall-signal-unit/).  I haven't used most of them but none look like they quite do what we want.  At NIST we usually just roll our own from a two-channel standalone DDS, a pile of Minicircuits bits and some custom passive rf filters and a custom active LPF on the output. 

Interface could be EEM and/or Ethernet.  Having no Ethernet will be obnoxious if we don't want to put an EEM master on Stabilizer.  Having ethernet could also make the board on option for people who aren't otherwise using Sinara - the fact there are quite a few commercial products out there demonstrates there is at least some market.

dtcallcock, 2019-06-02:
Just to finish answering @gkasprow questions:

> Low pass filter series: LFCN-3000+

Yes, the Minicircuits FV1206 package used in this part should be fine for DDS band defining filters.  How about [LFCN-160+](https://www.minicircuits.com/pdfs/LFCN-160+.pdf) (-1.5dB @205MHz, -19dB@300MHz) as the default population?

> Fixed attenuators: YAT-12+

MCL YAT series sounds fine.

> modulators: ADL5375-05ACPZ-R7, ADL5380ACPZ-R7

I think we just want a double-balanced mixer rather that an AD part. Will open another thread on this.

> Low noise generators: CCHD-950-25-100.000, Si570, Si549

See thread on DDS clocking.

dtcallcock, 2019-06-02:
> what are the second two dds channels for?

For the other PDH channel that's a copy of the first.  Reasoning from above:

- It isn't much extra work (just copy and paste the analog section) and will probably add <100 of $ to the board cost.
- Stabilizer is 2-channel
- A lot of atomic references also have an AOM in the pump beam to offset it. You could use one of the two LO outputs for this instead of a second PDH lock. My guess it that this is why Qubig's box has an 'aux LO' output.

gkasprow, 2019-09-21:
Here are more-or-less finished schematics. CPU IO assignment is missing.
[Pounder.PDF](https://github.com/sinara-hw/Pounder/files/3639019/Pounder.PDF)

closed 2019-11-30 by gkasprow