sinara-hw/Pounder #2: DDS Clocking

dtcallcock, 2019-06-02:
How about using the same 100MHz oscillator ([CCHD-950-25-100.00](http://www.crystek.com/crystal/spec-sheets/clock/CCHD-950.pdf)) and MUX ([Si53340-B](https://www.silabs.com/documents/public/data-sheets/si5334x-datasheet.pdf)) as Urukul to allow selection between XTAL and ext ref?

Only issue is that Si53340-B has LVDS outputs and AD9959 datasheet only talks about connecting LVPECL and PECL clocks.

Ext ref could be a front panel SMA or internal MMCX depending on front panel space and perceived likelihood people will use it.

To answer @hartytp questions:

> what’s the use-case for the ext ref? Synchronised locks? 

Yes.  But also:
- In some kinds of modulation schemes (like ones with an offset AO) the DDS output gets added to the laser frequency and so you might want that frequency locked to a reference.
- The phase noise of the DDS is improved by bypassing the internal multiplier.  An ext ref would allow people to provide their own low phase noise 500MHz reference.

> If so is it worth having a ref out port?

Sure.  One of the other outputs of the MUX could be taken to an MMCX.

gkasprow, 2019-06-02:
The AD9959 clock input requires min 400mVpp input clock.
The Si53340-B specification says that it delivers 200mV..490mV single ended swing which means min 400mV pp. So that should work. If we route 120Ohm tracks and place 120Ohm load than we would get sth like additional 20% margin. But IMHO we already are well within spec.

dnadlinger, 2019-07-17:
> what’s the use-case for the ext ref? Synchronised locks?

For fibre noise cancellation (#7), one would definitely want to use a good external reference as well.

gkasprow, 2019-09-21:
copied directly from Urukul, added SMA input isolation
[image]

dtcallcock, 2019-09-21:
Can Q1 go to an MMCX so that the 100MHz internal oscillator is available if needed?

> added SMA input isolation

So EXT_CLK IN is a front panel SMA?  if so, what are the CK MMCX annotations next to CLK1 input on IC24?

gkasprow, 2019-09-21:
It is reminiscent from Urukul. Yes, it is SMA. OK for internal MMCX

dtcallcock, 2019-10-08:
Closing as new issues opened for remaining issues.

closed 2019-10-08 by dtcallcock