sinara-hw/Pounder #3: Power Amp

dtcallcock, 2019-06-02:
An output amp that can do more that +10dBm would be useful as often with a PDH you are often driving a modulator like an EO.  I see three options:

- Add [TSS-13HLN+](https://www.minicircuits.com/pdfs/TSS-13HLN+.pdf) that can be switched in and out of output as described in [main thread](https://github.com/sinara-hw/Pounder/issues/1).  The [HMC1118](https://www.analog.com/media/en/technical-documentation/data-sheets/HMC1118.pdf) switch looks like it will be able to handle the power.  Power dissipation is 2W/channel and it might add ~$50 to the BoM.  

- Leave room for power amps but make them DNP by default.  In that case I'd probably use the [PHA-13LN+](https://www.minicircuits.com/pdfs/PHA-13LN+.pdf).  It's a little less power (+23dBm) but it's an easily hand-solderable package and runs off 5V so wouldn't need dedicated rails populating (current draw is 160mA max per amp so presumably not that bad to build that margin in the existing 5V rail).  Lack of enable line also not an issue here.

- We could make a separate medium power RF amplifier board in Eurocard form factor (prob 2 ch).  Such a thing would be useful for other boards like Urukul/Mirny too - right now there is a bit of a gap in Sinara if you need RF power in the +10 to +25dBm range where Booster would be total overkill.

hartytp, 2019-06-02:
For AOs you’d usually need a bit more power than those wouldn’t you? The nice qubig EOs only need something like 13dBm. So is there enough of a use case for >16dBm <28dBm (which is prob more than you’d get after the switch) to be worth the hassle?

Personally I’d go for the biggest MMIC amp we’d be happy with (16dBm ish ideally) and let users stick an appropriate pre-packaged mcl amp on outside stabiliser if they need it (can always provide a header).

dtcallcock, 2019-06-04:
> For AOs you’d usually need a bit more power than those wouldn’t you?

Yes, but then using Booster would be reasonable.

> The nice qubig EOs only need something like 13dBm. 

It's true that for some cavity locks powers can be really low.  For atomic locks though we run them with as much as 23dBm.

> stick an appropriate pre-packaged mcl amp on outside stabiliser if they need it (can always provide a header).

Isn't this the kind of thing we are trying to move away from with Sinara?  Is option 2 above much harder than this - it's just a MMIC so only needs a handful of external passives around it.

hartytp, 2019-06-04:
>  Is option 2 above much harder than this - it's just a MMIC so only needs a handful of external passives around it.

It's not so hard. The chip eats 140mA at 5V (700mW) so you'll have to make sure you spec the circuit out for this. Will prob require a 5V5 SMPS on the AFE mezzanine to generate the supply for that from the main 12V rail. That may be necessary anyway, depending on how much current the other components draw from that rail.

If it's a bit of a niche use-case then solder jumpers sound like a good idea (maybe populate the amp by default but use a 0R resistor to cut off its supply to make life easier?). If it's going to be common then worth thinking about the switches more seriously

dtcallcock, 2019-06-04:
> If it's a bit of a niche use-case then solder jumpers sound like a good idea. If it's going to be common then worth thinking about the switches more seriously

Yeah, I think that given the switch has to be a bit extra-special then solder jumpers are the way forward.  

> maybe populate the amp by default but use a 0R resistor to cut off its supply to make life easier?

That would be my preference.  Adds $20 to the BoM is all.

hartytp, 2019-06-04:
Sounds good. If we find people aren't using it, we can always DNP it later on.

FWIW, I'd also be fine with connecting it to the output by default (I'm not so concerned with an extra 0.7W of power dissipation for this design as we won't have many channels of it).

gkasprow, 2019-09-21:
Attenuator schematic
[image]
power stage and switches
[image]

gkasprow, 2019-09-21:
What about using HC164 to drive the VGA?

gkasprow, 2019-09-21:
Or, I2C extenders would be simpler...

gkasprow, 2019-09-21:
Would it be OK from SW perspective to chain both attenuators serial interfaces and drive them from the same SPI?

dtcallcock, 2019-09-22:
If we go for the PHA-13LN+ amp (P1dB=+24dBm), which I think is probably the right choice, then we don't need those fancy HMC1118 switches (~$7 ea).  We can just re-use the [HMC349ALP4CE](https://www.analog.com/media/en/technical-documentation/data-sheets/HMC349ALP4CE.pdf) from Urukul. That switch is half the price and known-good.  The 0.1dB compression point is +31dBm and power handling is +33.5(+26.5) dBm on thru(terminated) path so it can handle the power.

The way the switches are connected right now, there isn't a way of switching off the output.  Obviously you could switch to the power amp route but leave the power amp powered down.  I don't know if that's safe or how much isolation it would give though.  On the other hand, the DDS has amplitude control so one can just set that to zero if desired.

dtcallcock, 2019-09-22:
The passives around the [PHA-13LN+](https://www.minicircuits.com/pdfs/PHA-13LN+.pdf) don't match the suggested datasheet values.  Also there is capacitor missing (C1 on datasheet) and the input capacitor and inductor (C2 and L2 on datasheet are switched).  Is there a reason for this?

dtcallcock, 2019-10-08:
> We could make a separate medium power RF amplifier board in Eurocard form factor (prob 2 ch). Such a thing would be useful for other boards like Urukul/Mirny too - right now there is a bit of a gap in Sinara if you need RF power in the +10 to +25dBm range where Booster would be total overkill.

It sounds like this might be worth revisiting if there isn't a suitable MCL amp. 

Otherwise closing.

closed 2019-10-08 by dtcallcock