sinara-hw/Pounder #4: Mixer

dtcallcock, 2019-06-02:
>>>> We use Minicircuits [SYPD-1+](https://www.minicircuits.com/pdfs/SYPD-1+.pdf) phase detectors btw. Does anyone have experience with something else?

>>> IIRC, there is nothing special about those phase detectors, they're just mixers. Any decent (good isolation/LO leakage) will do just fine.

>> Yep. The main reason we use them is that they are specified for phase detection so you know what performance you will get rather than having to infer or test - for example with a random mixer you may end up with some annoying drifty dc offset. Minicircuits have a nice application note on the subject.

> IIRC, the phase detectors are mainly specified with the RF input saturated, which is not what one generally wants for an atomic lock. With the RF input unsaturated there is less difference between a phase detector and a decent DBM. There is a good Marki app note on mixers as well. IIRC their conclusion is that isolation is a very good proxy for general mixer performance and, having done some really careful tests that use mixers as phase detectors, I've never had an issue with large offsets in mixers that have decent isolation/LO leakage etc.

> Anyway, if there is a phase detector mixer that covers the frequency band, power range etc that we need then sure, let's go for it. If not, then let's just use a decent mixer. I'm equally happy either way.

@hartytp Do you want to suggest a part number you have had good experience with or shall we go with the SYPD-1+?

hartytp, 2019-06-02:
I’ve done quite a lot of microwave interferometry recently and have generally found that any MCL DBM will be fine. The Marki app note is prob right that isolation is a good thing but usually the mcl/marki/etc mixers all have about the same specs. I can dig out the pn of the mixer we used in our home brew pdh box if you want, but personally I’d just go for anything with the right frequency range

dtcallcock, 2019-06-03:
> I can dig out the pn of the mixer we used in our home brew pdh box if you want, but personally I’d just go for anything with the right frequency range

You used an MCL [ADE-1L](https://www.minicircuits.com/pdfs/ADE-1L.pdf) but it doesn't quite go low enough in frequency.  It looks like the MCL [ADE-1+](https://www.minicircuits.com/pdfs/ADE-1+.pdf) will do the job though. 

Btw, here is another nice article on mixer selection: http://www.rfcafe.com/references/articles/wj-tech-notes/Mixers_phase_detectors.pdf

dnadlinger, 2019-07-17:
Measurements of DBM residual phase noise are hard to come by. There is a NIST paper with some results: Barnes et al., Residual PM Noise Evaluation of Radio Frequency Mixers (2011), [doi:10.1109/FCS.2011.5977868](https://ieeexplore.ieee.org/document/5977868). As Tom mentioned, most should work for our use case.

dnadlinger, 2019-07-18:
> the phase detectors are mainly specified with the RF input saturated

Just for the sake of completeness, for fibre nosie cancellation (#7), we'd want to use the mixer as a pure phase detector, i.e. well saturated.

gkasprow, 2019-09-20:
What's the conclusion then?
 Is MCL ADE-1+ fine?

dnadlinger, 2019-09-20:
The ADE-1+ looks like it should work. Let's just make sure that we have enough LO power available to saturate it (probably around 10 or 11 dBm), and to terminate the IF output into 50Ω at high frequencies (series RC to ground plus LRC for the DC path).

gkasprow, 2019-09-20:
OK, what about the LO amplifier? Would [ERA-2SM+](https://ww2.minicircuits.com/pdfs/ERA-2SM+.pdf) work here?
We also need VGA. What gain and frequency range do you need? Do you have some candidate?
Digitally controlled would be fine, but we have also two free DAC outputs from CPU.

gkasprow, 2019-09-20:
What about the coupler? Would [TCD-9-1WX+](https://ww2.minicircuits.com/pdfs/TCD-9-1WX+.pdf) already used in the Booster work for you?

dtcallcock, 2019-09-20:
> OK, what about the LO amplifier? Would ERA-2SM+ work here?

Looks good.

> What about the coupler? Would TCD-9-1WX+ already used in the Booster work for you?

No, that part is 5-2000 MHz.  I want to run the LO as low at 1MHz.

> We also need VGA. What gain and frequency range do you need? Do you have some candidate?
Digitally controlled would be fine, but we have also two free DAC outputs from CPU.

Where on the [block diagram](https://github.com/sinara-hw/Pounder/issues/1#issuecomment-497855600) would this VGA be?

dnadlinger, 2019-09-20:
> What about the coupler? Would TCD-9-1WX+ already used in the Booster work for you?

Not for the low-frequency locks @dtcallcock is targeting. The frequency range (1 MHz to 160 MHz) is a bit awkward in terms of directional couplers, though. Perhaps we can just use a tap as we don't need directionality? [ADCB-20-82+](https://ww2.minicircuits.com/pdfs/ADCB-20-82+.pdf) looks like it might be an option otherwise; I haven't had a look at other manufacturers parts.

gkasprow, 2019-09-21:
cannot we just use resistive power divider?

gkasprow, 2019-09-21:
There is a VGA driving the coupler

gkasprow, 2019-09-21:
I will use [LT5554](https://www.analog.com/media/en/technical-documentation/data-sheets/5554f.pdf)

gkasprow, 2019-09-21:
What cutoff crequency do we need at the mixer output?

gkasprow, 2019-09-21:
initial schematic
[image]

gkasprow, 2019-09-21:
I will add opamp lowpass filter at the output once i know the cutoff frequency

gkasprow, 2019-09-21:
power detector
[image]

dnadlinger, 2019-09-21:
To keep things organized, let's close this issues as we've decided on a mixer (for now, anyway).

@gkasparow: I assume ADPQ-2-250 is just a stand-in? Frequency range of course wouldn't work for us. For the DDS reconstruction filters posted elsewhere, 200 MHz like on the eval board would seem more appropriate, given that we are planning on using it at 160 MHz.

closed 2019-09-21 by dnadlinger

gkasprow, 2019-09-21:
Yes, I didn't change it while modifying other components. Of course, we will use ADCB-20-82+.
FL3 is LFCN-160.

dnadlinger, 2019-09-21:
For the splitter, a resistive tap might work just as well, I'll leave that for someone else to comment on.