sinara-hw/Pounder #5: Output filter

dtcallcock, 2019-06-02:
Filter to go between mixer output and Stabilizer.

In order to take advantage of a potential future FPGA-based fast Stabilizer we should have the option to make this pretty minimal:

> the bandwidth of the signal chain after downconversion should be roughly 10 MHz to not completely take up the phase margin of a 2–3-ish MHz feedback loop

>One usually has a band-stop filter (LPF costs too much phase margin) after the mixer to remove any LO leakage, which can otherwise get rectified by following stages, degrading the DC stability. Since we would like to support a wide range of LO frequencies, this will either need to be a fixed filter selected by users (probably the easiest option) or widely tunable. A SDR approach might be fun here. ie. a minimal fast AFE after the mixer and then rely on a fast ADC + DSP to do the filtering as well as the loop-filter

However for things like slow atomic locks a fairly aggressive LPF would be nice.  

Also need to decide how much gain is needed.

hartytp, 2019-06-02:
Probably best to do as much of the lpf in software as possible, does this lpf need to do any mare than remove rf leakage? Stabiliser actually already has this so nothing needed there, but maybe keep pads for something in case?

dtcallcock, 2019-06-03:
> does this lpf need to do any mare than remove rf leakage? Stabiliser actually already has this so nothing needed there

I don't think that's true - stabiliser has 1MHz of bandwidth but we want to be able to run the rf as low as that.  I also think it might be worth helping the digital filter out a bit by not making it average through 1MHz of noise bandwidth (and something like an Iodine lock is very noisy) for a 100Hz bandwidth lock.  

I think my preference would be to buffer the mixer with an op-amp circuit as shown on page 2 of the [MCL app note](https://www.minicircuits.com/app/AN41-001.pdf).  This circuit would have the full 10MHz of bandwidth but have component pads to allow this to be reduced to as low as say 10kHz.

dnadlinger, 2019-07-17:
Just to state the implicit assumptions: The mixer output filter should be a decent 50Ω termination at higher frequencies and provide some kΩ of load at DC (e.g. an AC-coupled 50Ω termination plus an LCR lowpass as in Figure 2 of the above paper).

I'd also use an op-amp with reasonable EMI rejection, just in case. A low-noise FET-input opamp should definitely work. The choice somewhat depends on the target bandwidth; e.g. OPA828 or ADA4625.

For use as a phase detector, we should only need a gain of a few.

dnadlinger, 2019-07-29:
The "above paper" should have been the one mentioned in #4: Barnes et al., Residual PM Noise Evaluation of Radio Frequency Mixers (2011), doi:10.1109/FCS.2011.5977868.

gkasprow, 2019-09-21:
Filter schematic
[image]

Filter response including component tollerances
[image]

hartytp, 2019-09-23:
@gkasprow For the output OpAmp supply, can we go for +-5V instead of +-12V?

My logic here is that we never need more than +-1V output for S/N reasons and I'd like this design to work well for fast applications (diode laser line width narrowing) where the total loop BW is a couple of MHz, so the filter should provide 10MHz of bandwidth to avoid degrading the phase margin too much.

+-1V is generally nicer for fast ADC front ends, and there are many more choices for fast OpAmps with 10V supply than 24V supply.

Aslo, can you set the AFE gain so that the output is approx +-1V (2Vpk-pk) when the mixer is saturated, and add an annotation about this please?

hartytp, 2019-09-23:
I'd also prefer to use a faster OpAmp to make it easy to increase the BW by changing passives if desired.

dtcallcock, 2019-10-08:
Closing as new, more specific, issues cover all the points above.

closed 2019-10-08 by dtcallcock