sinara-hw/Pounder #6: form-factor

hartytp, 2019-06-03:
Double checking: is everyone happy making this a stabilizer AFE mezzanine instead of a stand-alone eurocard? (Probably copy most of it directly from Stabilizer, including PoE and EEM connector).

Might be worth making a decision about how we want to do the fast servo first (https://github.com/sinara-hw/meta/issues/26?_pjax=%23js-repo-pjax-container#issuecomment-498192023). IME if the fast servo is going to be an EEM then this should probably be a standalone card. If the fast servo is going to be a stand-alone card (fast stabilizer) then we should make this an AFE.

dtcallcock, 2019-09-07:
If someone needs to use Pounder standalone then isn't that still possible if it's a Stabilizer AFE?  The Pounder outputs are automatically available on the stabilizer SMA 'input' connectors. 

The main disadvantage I see is that 'standalone Pounder' becomes 8HP instead of 4HP and would be slightly more expensive.  That seems like a worthwhile tradeoff for the board simplification and cost reduction though.

> Might be worth making a decision about how we want to do the fast servo first (sinara-hw/meta#26).

If the fast servo does digital PDH then this will be a moot point.

dtcallcock, 2019-09-22:
Re-open if you have second thoughts about current Stabilizer AFE plan.

closed 2019-09-22 by dtcallcock