sinara-hw/Pounder #7: Fibre noise cancellation using Pounder

dnadlinger, 2019-07-17:
Apart from various laser frequency locks, a common use case right up Pounder's alley would be fibre noise cancellation. There are few implications for the design beyond what we already have; I'll create separate issues/comments for them. This is the general overview/tracker issue on the topic.

For the uninitiated, the basic problem is that thermal and acoustic noise in optical fibres leads to drifts in the total path length that are significant below some kHz laser line width. By reflecting a portion of the transmitted light back through the fibre, (twice) the accumulated phase difference can be measured at the source end and hence compensated by applying an an appropriate phase shift before the fibre.

This is usually done by inserting an acousto-optical modulator before the fibre, driven at some frequency f_AOM, e.g. 80 MHz. The retro-reflected light passes the AOM again and is interfered with the unshifted light to create a beat note, which is mixed down using a high-quality LO at 2 * f_AOM and stabilised. The loop bandwidth is usually some tens of kHz, limited by the fibre (and AOM) propagation delays.

In the early literature ([1](https://www.osapublishing.org/ol/abstract.cfm?uri=ol-19-21-1777), [2](https://www.spiedigitallibrary.org/conference-proceedings-of-spie/2378/0000/Accurate-cancellation-to-milliHertz-levels-of-optical-phase-noise-due/10.1117/12.208231.short?SSO=1)) as well as in the Zürich TIQI group ([3](https://tiqi.ethz.ch/content/dam/ethz/special-interest/phys/quantum-electronics/tiqi-dam/documents/semester_theses/semesterthesis-nelson_darkwah_oppong.pdf)), this was typically done by driving the AOM using a VCO. (From what I know, similar systems exist at JILA/Innsbruck/… as well.) I don't see a reason why this couldn't be done much more conveniently using a DDS, though.

On Pounder, we would use one DDS to create a stable LO at 2 * f_AOM (e.g. 160 MHz). For this, we definitely want to be able to clock the DDS of a good external reference (and pay some attention to PLL quality), as this sets (a limit to) the stability of the optical frequency. The second DDS would run at half that frequency, f_AOM (e.g. 80 MHz), with frequency and phase offset constantly being updated by Stabilizer to implement a PI loop response.

Additionally, in many lab-local use cases it would be convenient to use the same AOM to also stabilise the optical power after the fibre (and polarisation cleanup) by feeding back onto the RF drive amplitude. For this, one would connect a photodiode to the second Stabilizer ADC channel, feeding back onto the DDS amplitude word.

The obvious choice would be to use two AD9910s and an FPGA, with the parallel interface wired up for one of them. Since we don't need a lot of loop bandwidth here, though, it looks like a single AD9955 should work too. Its SPI interface appears to be faster than that of the AD9910, up to 200 MHz, and 10 bit should (barely) be enough. A single update would be 72 bits, so using 100 MHz SPI (SDR) driven from the Stabilizer ARM core should get us > 1 MHz update rate. (There will be processing latency, … on top of that, of course.) There would be an advantage to using a pair of AD9910s, though, namely the higher frequency range – many commonly used AOMs in the blue are run at 200 MHz, so having 400 MHz in the first Nyquist zone would be somewhat convenient..

As discussed in #4, most decent mixers should be good enough not to be a limitation in terms of system phase noise performance. We definitely want to be able to saturate it well using the LO, though, that is, to use it as an AM-insensitive phase detector. (One could use a PFD just as well, but capture range isn't relevant here, as the average frequency will always be exactly half the known LO frequency.)

Features discussed elsewhere which are not relevant for this:
 - Output amplifier (there is going to be an AOM power amp anyway)
 - Second channel (the second ADC is taken up by intensity stabilisation)

dtcallcock, 2019-08-30:
> it looks like a single AD9955 should work to

Do you mean the AD9959 (I'm guessing this is a typo as AD9955 is long obsolete)?

>  There would be an advantage to using a pair of AD9910s, though, namely the higher frequency range – many commonly used AOMs in the blue are run at 200 MHz, so having 400 MHz in the first Nyquist zone would be somewhat convenient.

Do you think it's worth opening an issue to discuss DDS choice or are you happy to stick with AD9959?

gkasprow, 2019-09-20:
What about adding EEM connector to the Pounder? One can then connect it to Kasli and drive with 200MHz SPI.

dnadlinger, 2019-09-20:
> Do you think it's worth opening an issue to discuss DDS choice or are you happy to stick with AD9959?

I'm happy to stick with the AD9959 if we are going for a two (well, four) channel design. The extra BOM cost for 4xAD9910 over a single AD9959 seems a bit hefty for spectroscopy-type applications. For most noise-cancelled fibre links (674 nm/729 nm/…) 80 MHz AOMs are perfectly good anyway, and for other applications, we can spin a second board with 2xAD9910 down the line, cook something up using Kasli, or play around with harmonic downconversion.

gkasprow, 2019-09-20:
One can also connect external Urukul to Sampler when the higher frequency is needed - Sampler has downstream EEM port.

dtcallcock, 2019-09-20:
> What about adding EEM connector to the Pounder? 

This is a little non-trivial if we design the board with no front panel output, we are use >8 of the 16 DIO on the Stabilizer header and we use the Stabilizer CPU DAC and ADC.  All of these could be worked around if someone really thinks this is a good idea though.

gkasprow, 2019-09-21:
The DDS can work in SDIO x4 mode. At the moment, QSPI is routed to FLASH memory that is DNP right now. I can route it to the AFE header instead of general IO line. In this way we would use SPI1 for attenuator and QSPI for DDS.

dnadlinger, 2019-09-24:
I haven't checked the interfaces on either CPU or DDS end – maybe @jordens has an informed gut reaction here?

dtcallcock, 2019-10-08:
Closing as QSPI idea covered here: https://github.com/sinara-hw/Pounder/issues/10

@dnadlinger reopen if you want to discuss this use case further in light of other issues.

closed 2019-10-08 by dtcallcock