sinara-hw/Pounder #9: Digital -> analog crosstalk from DDS SPI updates

dnadlinger, 2019-07-17:
For PDH-type locks, the digital lines to the DDS(es) will mostly be quiet, but for fibre noise cancellation (#7), there will be continuous updates. We should be somewhat careful during layout to keep contamination to a minimum, particularly as one might end up with bus frequencies close to the RF/LO.

dnadlinger, 2019-07-17:
(The signal integrity on those lines of course must also be good enough to support high-speed operation of the serial bus.)

dtcallcock, 2019-09-07:
If Pounder is a Stabilizer AFE then this signal integrity requirement also applies to Stabilizer.

@gkasprow Do you think the GPIO header on Stabilizer is up to running a 100-200MHz SPI interface?

gkasprow, 2019-09-07:
The GPIO header should not be an issue, I'm not sure about the CPU. I can make the SPI properly terminated 50Ohm (or 75 ohms) and increase the track clearance to support such a mode.

gkasprow, 2019-09-21:
DDS schematic
[image]

jordens, 2019-10-12:
dds schematic looks good.

jordens, 2019-11-27:
Also ensure SPI CLK digital cleanliness, maybe even using the same approaches as on Fastino.

gkasprow, 2019-11-30:
I terminated all fast digital SDIO lines

closed 2019-11-30 by gkasprow