sinara-hw/Pounder #10: DDS control

gkasprow, 2019-09-21:
What shall I do with P3...P0, SYNC_IN, IO_Update pins? Shall I route them to ordinary IOs or there are some special wishes?
And what about using QSPI in STM32. It is optimized for FLASH, but one can disable addressing and transfer ordinary data as well.

gkasprow, 2019-09-29:
@jordens did you use STM32 QSPI FLASH interface in SDIO mode? AD9959 has such access mode.

jordens, 2019-09-30:
I haven't used the quadspi peripheral. Might work with that dds.

gkasprow, 2019-09-30:
Let's add 0R resistor option so we can switch to SPI in case of issues.

jordens, 2019-10-12:
The quadspi Periphareral is ok for this. Memory mapped access will be of very limited use because of the variable register length. For phase locking (fiber length stabilization is a pll) doing it on both channels would be tricky because of the way the target channel is switched. But that's a limitation of the dds, not the spi bus.

jordens, 2019-10-12:
* SYNC_IN should be a CPU timer channel output and SYNC_OUT should also go to a timer input. Or ignore both if you are short on pins.
* IO_UPDATE and P0-3 should go to GPIOs, preferrable within the same GPIO bank of the CPU.

gkasprow, 2019-11-30:
control signals assignment
[image]
[image]

closed 2019-11-30 by gkasprow