sinara-hw/Pounder #11: Schematic rc

hartytp, 2019-09-22:
@gkasprow can we have a pdf of the current schematic release?

gkasprow, 2019-09-29:
It's here 
[Pounder.PDF](https://github.com/sinara-hw/Pounder/files/3666655/Pounder.PDF)

hartytp, 2019-09-29:
Thanks @gkasprow !

I'll review this tonight.

To be clear though, I will only do a very high-level review. From experience, I've found it's essential that someone (other than Greg, since fresh eyes are necessary) does a detailed review, which I won't be able to do for this design. @dtcallcock will you do this? It's something that really shouldn't be a "learning project" for a student, since it needs someone whose made some mistakes before. Even if much of the design is copy-paste, it's still easy for bugs to creep in when bits are moved from one design to the other

FWIW, the things I typically look at include for every IC on the board:
- [ ] Abs max rating for all pins
- [ ] Pin descriptions
- [ ] reference designs/eval boards
- [ ] Necessary capacitances v(e.g. LDO stability), compensation components, etc
- [ ] Thermal management/power budgets and handling
- [ ] max currents
- [ ] skim of basic specs to look for potential isues
- [ ] voltage standards, etc
- [ ] quality of annotations
- [ ] check someone is looking carefully at the digital stuff (@jordens ?)

That's about it, on top of general sanity checking.

Anyway, sorry if that's stating the obvious, but I want to make sure it doesn't get overlooked, since my experience is definitely that skimping on design reviews wastes huge amounts of time

dtcallcock, 2019-09-29:
> @dtcallcock will you do this?

Yes, though I'm a bit swamped this week with the start of term and family visiting.

> check someone is looking carefully at the digital stuff (@jordens ?)

I'm looking into funding firmware development that will include making sure the right connections are there.

hartytp, 2019-09-29:
High-level review complete. Looks like a cool design once the details are sorted.

Looking forward to removing some more crappy electronics from my lab once this project works ;)

dtcallcock, 2019-10-08:
I've done a first-pass.  Probably makes sense to have a new version and a preliminary power budget before looking again @gkasprow?

gkasprow, 2019-11-30:
I will open new issue once rc is ready

closed 2019-11-30 by gkasprow