sinara-hw/Pounder #13: vga

hartytp, 2019-09-23:
Is 18dB max enough gain for the VGA? Would it be easier to just add the same digital attenuator on Urukul etc followed by a fixed-gain pre-amp?

dnadlinger, 2019-09-24:
Wouldn't the VGA typically be used only to fine-tune the gain after you've used an external amplifier (e.g. an amplified photodiode) to get you close enough? Sure, having to fiddle with external attenuators/multiple cascaded amps/etc. is annoying, but it seems like an 16 dB window is wide enough to easily hit when designing/choosing the detection chain.

Without having calculated actual numbers, it seems like 18 dB gain ought to give you decent performance at -20 dBm input power or something like that.

jordens, 2019-11-27:
I would also think that a step attenuator (that's already present on the BOM) plus a fixed amplifier (that's also present on the BOM) is better than the PGA. Wider range, no more useful step sizes. Maybe also less insertion loss and thus better overall noise figure than the PGA. Also saves at least one transformer.

jordens, 2019-11-28:
* If you go for the step attenuators, hook them up in daisy chain with the two existing attenuators, but use individual LE lines (4 LE, 1 CLK, 1 SDI, 1 SDO).
* No need to wire ATT_RSTn on any of the attenuators IMO.
* OSC_en, DDS_RESET, EXT_CLK_SEL and the attenuator lines can all go onto a IO extender (I2C or SPI). No need for CPU GPIO.
* Update to the current Stabilizer IO connector pinout.

gkasprow, 2019-11-30:
@hartytp are you happy with two attenuators followed by ERA-2SM+ instead of PGA?

hartytp, 2019-11-30:
I’m happy either way

hartytp, 2019-11-30:
@dtcallcock should comment however

dtcallcock, 2019-12-01:
I don't see a problem with this way of doing it.

closed 2019-12-01 by gkasprow

gkasprow, 2019-12-01:
@jordens I assume you don't plan to use hardware NSS pin because we will always transmit 48bits

jordens, 2019-12-01:
On the attenuators?

gkasprow, 2019-12-01:
yes

jordens, 2019-12-01:
Yes. We can always load all attenuators. Here it's fine since it's easy to track the state either in software or by doing the non destructive procedure we have for urukul initialisation.

jordens, 2019-12-01:
But why is it 6 attenuators?

gkasprow, 2019-12-01:
We have two in output state and two cascaded instead of VGA, But I don't understand why we need to cascade two of them.

hartytp, 2019-12-01:
I don't remember why we have two of them either. (I also can't remember why we decided to use the attenuator instead of the VGA)

jordens, 2019-12-02:
I don't get why we need the VGA. The VGA gives you 2-18 dB gain, while using the existing step attenuator+gain block would give you -8 to 22 dB gain. And it has a 6 dB better noise figure!

gkasprow, 2019-12-02:
To cover required -20..+10dBm range we would need to cascade two amplifiers to have 10dBm on the mixer input. I'm not sure we need that much because we can also increase the gain with opamp at the mixer output and work with let's say 0dBm.

jordens, 2019-12-02:
Yes. Absolutely. The power at the mixer input only needs to be raised sufficiently to not matter for the noise. Going very high into the mixer (10 dBm) even for low input powers (-30 dBm) also kills the overall dynamic range. If we accept 10 dBm max input then with a 31.5 dB attenuator, 3 dB overall insertion losses, and a 24 dB gain block, we are at 0 dBm into the mixer and can maintain that 0 dBm level all the way down to -21 dBm RF input. Below that we start loosing performance, but probably only because of the noise figure of the input. Nothing a good external preamplifier and filter wouldn't solve. Above 10 dBm we would start clipping in the opamps which may also be fine since we tend to work in quadrature, i.e. at 0 V mixer output. Later the gain block would saturate before everything goes up in flames.

hartytp, 2019-12-02:
> And it has a 6 dB better noise figure!

At low power levels. Obviously the NF is worse than -30dB at max power.

gkasprow, 2019-12-02:
to use a single-stage gain block after the attenuator, we can go for one of them
[TSS-13HLN+](https://www.minicircuits.com/WebStore/dashboard.html?model=TSS-13HLN%2B) It  has NF=1.4 but costs 10.7$
Popular old [MAR-8A+](https://www.minicircuits.com/WebStore/dashboard.html?model=MAR-8A%2B) is just 1.5$ but has NF=3.1

jordens, 2019-12-02:
No point in trying to beat the transformer+attenuator insertion loss with an expensive amplifer afterwards. ERA-3 or MAR-8 is all fine. MAR-8 might be preferable because it moves the input range down.

jordens, 2019-12-02:
> At low power levels. Obviously the NF is worse than -30dB at max power.

More like 20 dB if you allow more power at the mixer.
But does that matter?

hartytp, 2019-12-02:
> But does that matter?

Likely not in most cases.

gkasprow, 2019-12-02:
So, what's the conclusion? Does such a solution work for all?
[image]

gkasprow, 2019-12-02:
edit, power levels were wrong

gkasprow, 2019-12-02:
If the gain of MAR8 is too high, it can be easily replaced with ERA2. It has the same footprint and bias current. Gain gets reduced from 31 dB to 16dB

hartytp, 2019-12-02:
@gkasprow does altium give one a nice way of swapping MAR8 FOR ERA2 in the BOM as a variant without having to have both on the schematic?

gkasprow, 2019-12-02:
Yes, but it must have the same footprint and come from the same library. 
Old Altium could only add alternate value, the new AD can define component replacement.
variant1
[image]
Variant2
[image]

hartytp, 2019-12-02:
Aah, cool!

The reason I ask is that I'm thinking about making a high-frequency  (up to around 4GHz) version of Pounder with PLLs instead of DDSs. I assume this will be a new project that starts as a copy of Pounder, but was curious how much I might be able to do as a straight up variant...

gkasprow, 2019-12-02:
We still have some board area, but it would be probably easier to do as a separate project. Production would need to be done as a separate process anyway. Variant routing of 4GHz signal and shielding could be tricky.

gkasprow, 2019-12-02:
@hartytp do you want to make it like a dual-channel design?
If the rest remains the same, we can place synthesizer in the area under the shield in the rear board area
[image]
to support two channels, would you need two synthesizers?

hartytp, 2019-12-02:
I'm not overly fussed about one channel v 2. It probably makes sense to keep it as a new design since there will be a lot of changes (baluns, filters, mixer) and I'm not sure we'll be able to make everything FP compatible.