sinara-hw/Pounder #15: output RF isolation

hartytp, 2019-09-29:
I'd really like to see the RF outputs properly isolated using 1:1 RF transformers and insulating washers around the SMAs. 

This isn't just a theoretical concern. We often PDH lock our lasers by current modulating the laser and I've found that doing this without isolation transformers kills the laser performance. We lock quite a few of our IR lasers using Toptica PDH boxes, with current modulation, and in every case I've ended up having to add isolation transformers to suppress noise.

hartytp, 2019-09-29:
I'm less concerned about isolating the RF inputs and the clock input, since those tend to be less problematic (although, there's also no harm in doing it).

dtcallcock, 2019-10-08:
Sounds good.  

Add DNP pads to allow user to connect to GND if desired though?

jordens, 2019-11-28:
And make sure that the two options are sensibly implemented. Always connect shield to the xfrm tap and then optionally that shield net to GND. Isolation xfrms on the inputs (clock and RF) are very nice.

gkasprow, 2019-11-30:
would[ ADT1-1WT+](https://www.minicircuits.com/pdfs/ADT1-1WT.pdf) do the job? We use them at the DDS outputs

closed 2019-11-30 by gkasprow