sinara-hw/Pounder #16: bi-dir coupler

hartytp, 2019-09-29:
Why are you using a bi-directional coupler instead of a 3-port directional coupler when you only need a forward power measurement? I've found the three port devices (with internal termination) tend to work better.

dnadlinger, 2019-09-29:
This is probably the one coupler Mini-Circuits has that goes all the way down to 1 MHz. Just building a simple tap from discretes might be the way to go, though.

gkasprow, 2019-11-30:
we can also go for [RDBC-20-63+](https://www.minicircuits.com/pdfs/RBDC-20-63+.pdf) which works from DC.
But can also be realized in the same way using just one resistor :)
The only coupler from Minicircuits that works from 1MHz is the one I used.

jordens, 2019-11-30:
You can't have both low frequencies and directionality. If you ensure a good match and isolation after/by the amplifier than then a tap is fine.

gkasprow, 2019-11-30:
well, you can build [directional coupler](http://michaelgellis.tripod.com/Bi-directional_amp.pdf) using opamps that work from DC :)
However, in our case we need to isolate LO leaking to RF input so existing solution may be optimal.

jordens, 2019-11-30:
Yeah. I thought I had restricted that statement to passive.

jordens, 2019-11-30:
The amplifier is pretty directional. You could put the tap between the step attenuator and the amplifier.

gkasprow, 2019-11-30:
True, providing that we go for this option instead of PGIA.

gkasprow, 2019-12-01:
I added 450R in series with power detector. This will give 10x attenuation

closed 2019-12-01 by gkasprow