sinara-hw/Pounder #18: document input power levels

hartytp, 2019-09-29:
It would be great to document (on the top page ideally) the RF input power level required to lightly (P1dB) saturate the input mixer with min/max PGA gain.

hartytp, 2019-09-29:
More a software point than a hw point, but it would also be great to have LEDs to indicate low/high power levels.

dtcallcock, 2019-10-08:
Also required input power level on ext_clk.

gkasprow, 2019-11-30:
There are 6  front panel green LEDs connected to the I2C expander

gkasprow, 2019-12-01:
Do we really need a double attenuator in front of ERA-2SM amplifier that drives the mixer?
with single attenuator, we accept -5.5dBm....+25.5dBm power level while driving the amplifier close to IP1.

hartytp, 2019-12-01:
I assumed we’d want the gain/attenuation setup to be more in the range -30dBm to 0dBm but @dtcallcock / @dnadlinger should comment on the desired range

gkasprow, 2019-12-01:
we would have to use an amplifier with much higher gain.
Single attenuator already has such gain window.

hartytp, 2019-12-01:
Indeed. I don’t think two attenuators is a good idea

We also need to be careful over attenuating too much in order to avoid noise issues with attenuate/gain cascades

hartytp, 2019-12-01:
I guess that’s the advantage of the vga... still, depends on what noise level people are looking to achieve here

dnadlinger, 2019-12-01:
-30dBm to 0dBm or -20dBm to 10dBm sounds more like it, yes.

gkasprow, 2019-12-01:
So, we definitely need to go for VGA. Otherwise we would need attenuator + dual stage amplifier

gkasprow, 2019-12-01:
The VGA gives only 2dB to 18dB Gain Range. Optimally we would need attenuator and then VGA, but this would add one RF transformer that isolates the input.
We can also place the attenuator after the VGA. The VGA should handle 10dBm of input signal (IIP3 is 27dBm).

hartytp, 2019-12-01:
If we need that much dynamic range vga followed by attenuator and fixed gain amp sounds good. 10dBm max input sounds fine

jordens, 2019-12-02:
I don't think we need that much dynamic range. A step attenuator gives us 31.5 dB dynamic range already. Apart from needing one more transformer, the VGA has a pretty disappointing noise figure and less dynamic range compared to step attenuator plus era-3. And we can't reuse the software architecture that we need for the other step attenuators.

hartytp, 2019-12-22:
Summarising this issue:
- [ ] @dtcallcock can you specify which max/min input power you want to support
- [ ] document this choice on the schematic
- [ ] design the attenuator/amplifier combo as well as the output buffer amp gain around this choice

dtcallcock, 2019-12-22:
I thought we were happy with the design @gkasprow laid out in #13 

[image]

hartytp, 2019-12-29:
@dtcallcock okay, so you're happy with +12dBm being the max input, meaning drives the mixer at roughly its ~P1dB and gives approx +-1V at the Sampler input?

If so, @gkasprow are you happy that the amp isn't saturated, the mixer is driven appropriately and the filter gain is set correctly (and that this is all documented)?

If so, let's close.

dtcallcock, 2019-12-30:
Works for me.

closed 2019-12-30 by gkasprow