sinara-hw/Pounder #19: filter amps

hartytp, 2019-09-29:
Can we use faster OpAmps for the filter please? I'd like this design to support ~10MHz analog bandwidth to make sure we can do fast locks on laser diodes (>=2MHz closed-loop bandwidth).

gkasprow, 2019-11-30:
Does this solution make you happy @hartytp 
[image]

hartytp, 2019-12-22:
@dnadlinger what was your final thought about OpAmps and filters?

After giving it some thought, my preference would be:
- include DNPd pads for an LC LO notch filter (users populate based on their requirements)
- have a low-pass filter by default
- use a high-speed OpAmp as a buffer/amplifier with gain chosen to match the mixer output to Stabilizer's +-1V input range
- aim for closed-loop BW of the OpAmp buffer to be >10MHz to allow for few MHz loops without too much loss of phase margin due to the buffer
- don't bother with an active low-pass filter (in any case, Stabilizer has decent filtering which should do a decent job of carrier rejection)
- Use a low-noise/low-offset +-5V FET OpAmp for the buffer

Any objections?

hartytp, 2019-12-22:
Let's also aim to choose passives + OpAmp to minimize noise from the output stage

dtcallcock, 2019-12-22:
> Any objections?

Nope.

gkasprow, 2019-12-29:
[image]

hartytp, 2019-12-29:
Worth having a pad for a capacitor in parallel with L8A?

hartytp, 2019-12-29:
Looks about right, but it's worth simulating a couple of filters to make sure that we can get the appropriate component values in the packages you've chosen (ideally with tight tolerance LC components and without silly small/high values to avoid the need for tuning).

The thing I'd be most interested in would be a ~30MHz notch combined with a general low-pass response with a cut-off frequency around ~10MHz.

hartytp, 2019-12-29:
Hard to tell from the schematic you've drawn but is the idea to have the OpAmp with a G=2 to give +-1V output? If so it will have ~10MHz unity gain bandwidth, which is okay (a little more might be nice to minimize RF rectification given that the input filter needs a relatively high cut-off to allow closed-loop BWs of a few MHz).

hartytp, 2019-12-29:
Anyway, looks good to me other than that, so long as we can implement the filters that people want (@dtcallcock )

dnadlinger, 2019-12-29:
> a little more might be nice to minimize RF rectification

Assuming the JFET frontend is well-behaved, shouldn't there be pretty negligible RF rectification (especially with a notch to filter)? That is, doesn't the GBW limit only act to further suppress LO feedthrough (if there aren't any diode-like funnies going on in the input stage)?

hartytp, 2019-12-29:
I was wondering that too. I admit that I don't know the answer since I don't have a full model of the OpAmp in my head. As I understood it, even JFET OpAmps often start to get non-linear around the unity-gain point and so are prone to rectification.

Also, we're only notching at f_lo, not harmonics. I haven't looked at the mixer data sheet, so not sure how much power is in those (they may be taken out by a general low-pass filter, but depends on the various frequencies involved).

Anyway, I'm happy to concede that I haven't given this too much thought and the current design could be fine (so long as we can get the LCs we need in the packages Gerg's chosen).

gkasprow, 2019-12-29:
I used 0805 and 0603 footprints so there are many RLC options possible.

gkasprow, 2019-12-29:
JFETs have two diodes between inputs and go non-linear for signals which exceed opamp SR.

gkasprow, 2019-12-29:
To avoid rectification just add simple RC filter with let's say 20MHz BW.

hartytp, 2019-12-29:
@gkasprow okay. LGTM then.

FWIW I would consider a [THS4631[(http://www.ti.com/lit/ds/symlink/ths4631.pdf) as:
- quite a bit cheaper
- nice low noise JFE amp, low offset etc (although not quite as low as the AD8610)
- higher GBP which reduces potential issues with rectification.
- It does have a higher IQ, but nothing I'm particularly fussed about for this design, particularly given we have +-5V rails

Other than that:
- [ ] I'd add a DNPd cap in parallel with L8A
- [ ] Add a DNPd cap in parallel with R13A (for simple low-pass)

And call it quits unless anyone has any other comments/suggestions?

Can you post a design RC when you get the chance?

hartytp, 2019-12-29:
Well, the plus I can see for the AD8610 is the extremely low offset (absolute and temp co). Not clear to me which will give better performance in the field...

gkasprow, 2019-12-30:
I don't like THS4631. It has a strange input circuit that works only with low-level signals, i.e. in TIA applications. Once I used it as a CCD sensor buffer but it didn't work well with a 0.5V input signal. It has some kind of protection that increases leakage current orders of magnitude for higher voltages. I also tried it in switched analog integrator application but it did not behave as expected. Finally, for CCD I used matched JFET + bipolar opamp and for the integrator changed the conception because I could not find that fast JFET amp.

hartytp, 2019-12-30:
Okay. I guess it's the slew rate enhancement circuit you're talking about?

I've used that part in other designs without issue, but I'm happy to avoid it if you've had problems.

I don't like the AD8610 as it's expensive for what it is and a bit slow. If I have time I'll post some other suggestions but this shouldn't delay production IMHO.

Let's leave this for now and move onto the next RC.

closed 2019-12-30 by hartytp

gkasprow, 2019-12-30:
Let's use THS4631 here. I've just noticed how expensive AD8610 is. It works in standard amplifier application, nothing switching. In case of issues, we can change to AD8610.

hartytp, 2019-12-30:
That sounds good. There are some other nice LTC JFET amps that are also cheaper and faster than the 8610. They're not as fast as the THS part but have somewhat lower power consumption (but not quite by a factor of two). Ultimately it depends on what we want to optimize for. Anyway, this isn't a big decision as any of these parts should work.

BTW last time I used the THS was actually for the phase shifter in our in-house (analog) PDH system. It worked well for an RF OpAmp phase shifter designed by a grad student ;)

gkasprow, 2019-12-30:
Other FET OAs
[image]

hartytp, 2019-12-30:
OPA657 is also fine. I think I discounted it before because the input offset voltage is a bit worse, but we can live with it. I don't see any advantage over the THS however (it also has a worse slew rate). The 657 is just a decompensated 656. The 627 is a bit too slow really and has a pretty high quiescent current for that speed (typical of 7mA).

Out of those, I'd stick with the THS.

hartytp, 2019-12-30:
Well, actually, the 656 is generally nicer than the THS. But it takes up even more quiescent current. at ~12mA per amp it's starting to actually be quite a bit of power (e.g. the +5V supply is generated from +7.5V on pounder).

hartytp, 2019-12-30:
656 is also quite a bit more expensive.

hartytp, 2019-12-30:
@gkasprow AD8065?

hartytp, 2019-12-30:
Well, these designs always involve compromise and there is no right answer. So, I'm happy with any of those.

I'd probably go for the AD8065 on consideration, largely driven by the lower power consumption and low cost. NB it's not short-circuit protected so make sure to add a decent output protection resistor.

Anyway that's all I have to say on the subject. I'll leave it to @gkasprow to make the final call...

dnadlinger, 2019-12-30:
Typical EMIRR+ for precision JFET op-amps (e.g. [OPA828](http://www.ti.com/lit/ds/symlink/opa828.pdf), [AD4625-1](https://www.analog.com/media/en/technical-documentation/data-sheets/ADA4625-1.pdf)) at tens of MHz is 40 dB. As expected, this is quadratic in voltage for FETs and [referenced to 100 mV](http://www.ti.com/lit/an/sboa128a/sboa128a.pdf), so 100 mV RF feedthrough would give ~1 mV offset, which is substantial.

Really depends on the input filter design, of course.

hartytp, 2019-12-30:
100mV of input EMI sounds like a lot however. That won't come from mixer feed-through. If the external EMI is that bad, users should be able to use an external (coaxial) BPF to remove it.

tl;dr I don't think this will be an issue but an AD8065 or similar seems like a better choice than the OpAmp that's currently there. But, I don't want to get stuck talking about this, so happy to defer to anyone else who has a strong opinion.

gkasprow, 2019-12-30:
There is 9k1 resistor which makes LPF with opamp capacitance (3.9pF)