sinara-hw/Pounder #20: IF gain

hartytp, 2019-09-29:
Let's make sure that saturating the mixer corresponds to +-1V output. That fits in nicely with the voltage range we're likely to have for the fast Stabilizer card -- +-10V is not necessary for S/N reasons and is a bit aggressive for high-bandwidth locks.

Let's also make sure that the output range is documented.

gkasprow, 2019-11-30:
@hartytp what levels can we expect at the output of such a diode mixer? Theoretically, it can be as high as the input signal, so it depends on the PGA setting and load resistor. I adde gain setting resistor to the second opamp.

hartytp, 2019-12-01:
For a 50 ohm load it’s just the conversion loss from the data sheet minus any compression isn’t it?

hartytp, 2019-12-01:
i vote we put something sensible here based on data sheet numbers and then test when the hw arrives and tweak as necessary for the v1.1 he

gkasprow, 2019-12-01:
ERA-2SM+ has 11dBm IP1 so if we drive the mixer with 10dBm minus 5dB conversion loss of the mixer we get 5dBm. This is 0.56Vp so we need roughly 2x gain to match the Stabilizer AFE range.

closed 2019-12-01 by gkasprow