sinara-hw/Pounder #21: IF notch filter

hartytp, 2019-09-29:
For higher bandwidth loops, a low-pass filter is often not a good option, since there isn't enough separation between the LO and the passband.

Can we have enough pads to allow us to have a passive RC low-pass filter followed by a single OpAmp notch filter please?

Let's aim to make the OpAmp fast enough to implement a 40MHz low pass filter.

gkasprow, 2019-11-30:
would such schematic implement all feasible filter schemes?
[image]

hartytp, 2019-12-01:
Thinking about this more, I think that having an active filter here is probably not necessary since we already have one on Stabilizer.

- Let's keep it simple and just have a fixed gain stage to match the mixer output to the +-1V range on Stabilizer
- Let's make sure that the OpAmp is fast enough to give >=30MHz unity gain bandwidth so that we we don't get too much phase lag at 3MHz
- Let's keep the pads to allow us to have an LC not filter to remove the carrier. Is it worth also having a second notch for the 3rd harmonic? LC filters should be DNP by default, use a coil craft package with a decent range of inductors so we can make notches for carriers up to ~50MHz
- This would allow us to e.g. lock a PZT via Stabilizer's output, and also take pounder's fast output (via the ADC SMA on Stabilizer) and hook that up to a fast lock (even just a simple P onto the laser current)

hartytp, 2019-12-22:
This is now basically a duplicate of #19

closed 2019-12-22 by hartytp