sinara-hw/Pounder #22: output nosie

hartytp, 2019-09-29:
FWIW the current design looks sub-optimal from a noise perspective. Maybe that's okay since we expect detector noise to always dominate? But, worth thinking about OpAmp/Johnson noise and making sure this doesn't accidentally limit system performance.

While we're on it, worth checking the usual things like bias-current offsets, self-heating effects on offset stability etc.

dnadlinger, 2019-10-06:
Regarding op-amps, I had suggested the OPA828/ADA4625, but that was under the assumption that
 - the target is ±10 V out (for which there isn't a good reason for fast loops),
 - we want a non-BJT input stage to hot have to worry about DC offset drifts due to rectified RF,
 - 1/f noise and offset drifts are worth thinking about, and
 - we are using a kΩ-level load on the mixer (which is good for performance when used as a saturated phase detector for fibre noise cancellation, but might not be appropriate for 10 MHz IF bandwidth).

There are probably better choices for a more general, wide-band demodulator.

hartytp, 2019-10-06:
Before finalizng the OpAmp choice, it would be worth simulating a LO notch filter for a realistic max frequency. Say, 40MHz (much higher than that and simple passive filtering will do the job)?

I suspect that something like an AD8021 would do a good job, but needs checking.

dtcallcock, 2019-10-08:
Can I leave it to one of you guys to pick something as I'm not as up on the latest opamp fashions and trends.

gkasprow, 2019-11-30:
So, what type of the filter you want me to simulate? Do you really want to build 40MHz filter using opamps?

hartytp, 2019-12-01:
> So, what type of the filter you want me to simulate? Do you really want to build 40MHz filter using opamps?

No

hartytp, 2019-12-01:
A fast FET amp sounds good here.

closed 2019-12-22 by hartytp