sinara-hw/Pounder #24: power amp

hartytp, 2019-09-29:
Do we really need the power amp? Is this a niche use-case better served by a packaged amp near the load?

If we do need it, someone needs to check the design carefully. Power levels appropriate, power handling of switches etc, thermal management, etc. Also make sure the PA is powered down by default. Maybe have an LED to indicate when it's active?

hartytp, 2019-09-29:
NB any coupling between the RF output and input can give offsets on the lock signal. Since we have a lot of gain here, we'll have to be very careful about decoupling the two halves of the circuit.

gkasprow, 2019-09-29:
Shall I add a shielded cage for Rx part?

dnadlinger, 2019-10-06:
I've opened #27 regarding shielding.

dtcallcock, 2019-10-08:
Ok, let's scrap this.

gkasprow, 2019-11-30:
I removed the power amp. it is still on the schematics but disabled with compile mask so you can still change your mind

closed 2019-11-30 by gkasprow