sinara-hw/Pounder #27: Shielding and RF out/in crosstalk

dnadlinger, 2019-10-06:
As pointed out in #24, with Pounder we have a coherent receiver on the same PCB as (potentially) the sender power amp, so avoiding crosstalk from the RF (LO) output to the input signal path is a concern – we care about DC stability for this. Shielding from external EMI is also a concern.

Having previously been bitten by this (trap RF coupling into PDH locks of nearby frequency), I would probably tend to go overboard here in typical physicist-who-has-opened-expensive-test-equipment fashion. Unless someone has a certain spec in mind here, though, perhaps the way to go here is simply to pay attention to separation/compactness in layout, and add provisions for a shielding can for the input side of things.

dtcallcock, 2019-10-08:
> Unless someone has a certain spec in mind here, though, perhaps the way to go here is simply to pay attention to separation/compactness in layout, and add provisions for a shielding can for the input side of things.

I like this plan.

dhslichter, 2019-10-10:
I would argue that a shielding can on the *output* side would also be beneficial.  Since we are talking about frequencies above a few MHz, cans can be pretty effective, and don't have to be that expensive either.  Another "old fashioned" way to do this is to put the input and the ouput on opposite sides of the board (if there is space).  You do have to ensure then that you are choosing different lock frequencies for neighboring Pounder cards.

gkasprow, 2019-11-30:
I added cages but we have very little space so it's possible that only Rx part would be shielded.

closed 2019-11-30 by gkasprow