sinara-hw/Pounder #28: Mixer IF load

dnadlinger, 2019-10-06:
The mixer IF output is currently 50Ω-terminated at AC, with a corner frequency of ~ 1 kHz, and 4.7kΩ at DC:

[image]

I guess this might have been a misunderstanding based on the diplexed setup I had suggested elsewhere for optimising the voltage/phase slope at DC (for which a more appropriate corner frequency would be a MHz or so). 

Perhaps we should just go with a simple 50 Ω load by default, though, as this is definitely the most universal (if slightly suboptimal) option across all use cases, and should avoid confusion.

dtcallcock, 2019-10-06:
The load @dnadlinger suggested from https://doi.org/10.1109/FCS.2011.5977868 was:

[image]

This is based on [this paper](https://doi.org/10.1109/FREQ.1977.200168) and [this one](https://arxiv.org/abs/physics/0608211).

dtcallcock, 2019-10-06:
> Perhaps we should just go with a simple 50 Ω load by default

I think this is a reasonable approach. We should have the pads to allow people to realize the above topology though.

dnadlinger, 2019-10-06:
@dtcallcock: Thanks for collecting the links. Yes, having the pads is probably a good idea. If we include two parallel pairs of pads for a series-Z connection to ground (like the RC in the above), people could also put on passive notch filters at f/2f like in the old PDH literature. (I've never tried to measure whether that's actually a good idea.)

gkasprow, 2019-11-30:
I implemented 50R by default. The topology from the paper can be implemented as well.

gkasprow, 2019-11-30:
[image]

closed 2019-12-01 by gkasprow