sinara-hw/Pounder #30: Stabilizer firmware support

dtcallcock, 2019-10-06:
A bare-bones set of ethernet commands to run Pounder may be:
- Set PGA gain
- Read CPU ADC
- Read power detectors using CPU ADC and convert to power (assuming datasheet specified component behavior - doesn’t need to be calibrated)
- Select oscillator input for DDS
- Enable clock multiplier with 4x, 5x, or 50x (for using 100MHz onboard oscillator or 10/100/125MHz ext ref)
- Set digital attenuators
- Set the RF output switches
- Write general DDS register
- For each DDS channel:
	- Enable/disable output
	- Set amplitude
	- Set phase
	- Set freq

Anything else that would be needed off the bat?

jordens, 2019-10-12:
* LEDs
* Readback of DDS register, attenuator state

hartytp, 2019-12-01:
Not a hw issue

closed 2019-12-01 by hartytp

jbqubit, 2019-12-01:
Updated [wiki ](https://github.com/sinara-hw/Pounder/wiki) to reference repository for firmware.