sinara-hw/Pounder #32: DC block caps

dtcallcock, 2019-10-07:
- 2.2uF caps are used everywhere except for C26 (10nF).
- What part number are these?  I could imagine some general purpose 2.2uF ceramics having pretty poor RF performance at 200 MHz.
- The section between C13 and C25 is floating.  Can one of these be default populated by 0R?

dtcallcock, 2019-10-08:
[ATC 530L] are 100nF broadband DC block caps specced from 16kHz to 18GHz.  A little bit pricey ($0.5) but guaranteed to work nicely.

dtcallcock, 2019-10-08:
C95 also needs to be the same type as our agreed DC block cap.

jordens, 2019-11-27:
And the ACG caps on the attenuators also need to be the same as the signal path ones.

jordens, 2019-11-27:
IMO 2.2uF is really large. Going that low in frequency (a) competes with what Stabilizer could do in baseband without downconversion, certainly for the LO paths, (b) makes properly biasing the amplifiers harder, and (c) risks having high ESL due to design and the large capacitance-to-volume ratio and correspondingly worsening the high frequency performance above the series resonance.

gkasprow, 2019-11-30:
we can use two caps connected in parallel: 10nF and 2.2u.
with 1nF would be hard to go down to a single MHz range. 10nF at 1MHz is 16Ohm.
If we go for these wideband caps (but their availability could be an issue) we still get Z=1.6Ohm

hartytp, 2019-11-30:
I always worry about resonances when paralleling cape. Might be worth simulating first

gkasprow, 2019-11-30:
the ATC530 is available everywhere. One must skip ATC and simply type 530L104KT16T 
Let's use them.

gkasprow, 2019-11-30:
I even have them in the libs :)

closed 2019-11-30 by gkasprow

jordens, 2019-11-30:
If you really want to go to those low DDS frequencies, some thought should also go into biasing chokes and decoupling and potential crosstalk through that.