sinara-hw/Pounder #33: ERA3SM (output amp)

dtcallcock, 2019-10-07:
- Power annotation just before amp is wrong (should be -15dBm).
- We are using 100R to bias but [datasheet](https://www.minicircuits.com/pdfs/ERA-3SM+.pdf) suggests 107R as optimum for 7V operation.

gkasprow, 2019-11-30:
done
[image]

gkasprow, 2019-11-30:
@dtcallcock there was misleading "9dB" text on the 1dB attenuator symbol.
So we have -5dBm power level. Together with 23dB of gain it gives 13 dBm at the output. So T pad should have 4dB so we get 13dBm at the ERA-3SM+ output.

gkasprow, 2019-11-30:
We can connect two 2dB attenuators to save one BOM line.

gkasprow, 2019-11-30:
[image]

gkasprow, 2019-11-30:
since we decided to scrap power amp, this is not an issue anymore

closed 2019-11-30 by gkasprow