sinara-hw/Pounder #40: DDS output filter

dtcallcock, 2019-10-08:
Right now specified as DC to 160MHz.

Presumably this is an [LFCN-160](https://www.minicircuits.com/pdfs/LFCN-160+.pdf)?

[image]

An alternative part would be [ULP-216+](https://www.minicircuits.com/pdfs/ULP-216+.pdf).

[image]

Would give the same bandwidth but quite a bit better attenuation of the image frequencies.  More expensive and much larger footprint though so perhaps not worth it.

jordens, 2019-11-27:
Not worth it IMO. Have the user clean inputs and outputs with custom bandpasses if there is are problems with aliases.

gkasprow, 2019-11-30:
Yes, it is LFCN-160+ now. Before there was just a placeholder.

closed 2019-11-30 by gkasprow