sinara-hw/Pounder #41: AD8363 VTGT

dtcallcock, 2019-10-08:
The resistive divider sets this to 1.33V currently.  

From the [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/AD8363.pdf):

"Most of the characterization information in this data sheet was collected at VTGT = 1.4 V. Voltages higher and lower than this can be used; however, doing so increases or decreases the gain at the internal squaring cell, which results in a corresponding increase or decrease in intercept. This in turn affects the sensitivity and the usable measurement range. Because the gain of the squaring cell varies with temperature, oscillations or a loss in measurement range can result. For these reasons, do not reduce VTGT below 1.3 V."

Unless this was intentional, I think we should just stick with 1.4V

dtcallcock, 2019-10-08:
> The minimum appropriate CHPF for a given high-pass pole frequency is
> 
> CHPF = 29.2 × 10−6/FHPPOLE − 25 pF 
> 
> where FHPPOLE is in hertz.
> 
> The subtraction of 25 pF is a result of the on-chip 25 pF capacitor in parallel with the external CHPF. Typically, choose CHPF to give a pole (3 dB corner) at least 1 decade below the desired signal frequency. Note that the high pass corner of the offset compensation system is approximately 1 MHz without an external CHPF; therefore, adding an external capacitor lowers the corner frequency.

If we want to go down to 1MHz, then we want a 100kHz pole.  Therefore 270pF would seem more appropriate than the current 0pF (DNP)

gkasprow, 2019-11-30:
done

closed 2019-11-30 by gkasprow