sinara-hw/Pounder #42: harmonize transformers

jordens, 2019-11-27:
* T2-1T+
* ADT1-1WT+
* ADT1-1+

Maybe this can be simplified. I'd consider going for the cheaper/simpler T1-1xx style everywhere. And maybe the T2-1 things won't be needed at all if the PGA becomes a step attenuator plus fixed amplifier.

On TR1 I also think it's wrong (has 50R diff termination but 2:1 ratio from a 50R SE source).

gkasprow, 2019-11-30:
So far we have:
4x ADT1-1WT+ at the DDS outputs
ADT2-1T+ as a clock input isolator. The ADT1-1WT+ should work even though it is 75Ohm
ADT1-1WT+ at the RF output
ADT2-1T+ at the PGIA output
ADT1-1+ replaced by TC2-1TX+ at the PGIA input

hartytp, 2019-12-22:
cf phaser balun choices

gkasprow, 2019-12-29:
After some optimization we have
DDS output, RF output:  ADT1-1WT+, 75 Ohms 0.4 to 800 MHz 1:1 RF Transformer
Ext CLK IN:  ADT2-1T+, 50 Ohms 0.4 To 450MHz 1:2 RF Transformer
FB IN:  TC2-1TX+, 50 Ohms 3 to 300MHz 1:1 RF Transformer
Can we replace the clock input transformer with  ADT1-1WT+ ? I would terminate it with 50Ohm instead of 100R

jordens, 2019-12-29:
fine with me.

hartytp, 2019-12-29:
> Can we replace the clock input transformer with ADT1-1WT+ ? I would terminate it with 50Ohm instead of 100R

Fine by me. You loose a bit on signal amplitude but gain on the bw which is equally important assuming the clock is driven by a square-wave source.

dtcallcock, 2020-01-05:
> FB IN: TC2-1TX+, 50 Ohms 3 to 300MHz 1:1 RF Transformer

There is no data on how this will work down at 1MHz. Do we expect this to be a problem?  Is there a reason not to use [TC1-1TX+](https://www.minicircuits.com/pdfs/TC1-1TX+.pdf).  It's the same package but 0.4-500MHz.

gkasprow, 2020-01-05:
changed to TC1-1TX+

closed 2020-01-05 by gkasprow