sinara-hw/Pounder #43: power input

jordens, 2019-11-27:
Why is there a separate power input again? Can't Stabilizer power this with its ~20W available?

hartytp, 2019-11-27:
I’ve found it useful in other designs to have a fp barrel connector, but there isn’t room on stabilizer. So this can be used to power both stabilizer and the afe

jordens, 2019-11-27:
Can you elaborate why there is no easy way to power Stabilizer from its rear power connectors or EEM in these circumstances?
And is it reasonable to tailor the design of Pounder so that Stabilizer can be powered?
#34 for front panel problems with 5 SMA and one barrel connector

hartytp, 2019-11-27:
In some of our racks the rear of Stabilizer isn't easily accessible and powering from an EEM/PoE isn't convenient. 

> And is it reasonable to tailor the design of Pounder so that Stabilizer can be powered?

I wouldn't call this tailoring the design. Point is just that if there is space on the FP so we can easily add the connector then it's a nice convenience feature (this is what we did on our current sense AFE and it's been useful). It there are good reasons not to do it (e.g. lack of space due to all those SMAs) then it's fine to drop it. It's not a big deal either way.

jordens, 2019-11-27:
Let's remove the power connectors from the back and make optimal use of the front panel real estate.

gkasprow, 2019-11-30:
@jordens The Stabilizer already have rear power connector, so there is no need to have it on Pounder as well.
To fit DC jack to the front panel, we would have to switch to MCX connectors, at least for the external clock input.

jordens, 2019-11-30:
I'd be a bit worried for ultimate phase stability with rotating connectors like mcx. But I don't have any data. It's a bit too much connector type diffusion for me. My choice would be to leave this all sma and omit front panel power in favor of the two existing alternative powering options.

hartytp, 2019-11-30:
I’m fine with that

gkasprow, 2019-11-30:
OK, done

closed 2019-11-30 by gkasprow

gkasprow, 2019-11-30:
this is proposed connector layout
[image]