sinara-hw/Pounder #44: review

hartytp, 2019-12-22:
@gkasprow @dtcallcock what's the status of this project? Are we waiting for a review by @dtcallcock, or are we waiting or @gkasprow to finish updating the project? I've lost track...

hartytp, 2019-12-22:
okay, I think almost all issues have been implemented. We need clarity from someone on what input power ranges we want to support, but other than that it looks like we should be able to produce another schematic RC relatively quickly from here.

dtcallcock, 2019-12-22:
I was waiting on a new schematic after that slew of issue activity about a month ago.  Let us know if there is anything you still want to discuss though @gkasprow

gkasprow, 2019-12-22:
It was on hold due to Booster and  Phaser. We will finish it after Xmass

hartytp, 2019-12-29:
Nothing to do here. Let's open a new issue when there is a new RC that fixes all current issues...

closed 2019-12-29 by hartytp