sinara-hw/Pounder #45: rc0 review

gkasprow, 2019-12-30:


gkasprow, 2019-12-30:
@hartytp the newest release is [here](https://github.com/sinara-hw/Pounder/releases/tag/v1rc0)

hartytp, 2019-12-31:
- [ ] what are the components in grey? e.g. the PGA? Are those a population variant? Can we just delete them since I don't see us ever wanting them and I'd like to avoid filling the design up with kruft

hartytp, 2019-12-31:
the OpAmps draw a decent amount of current, which comes directly from the P/N 12V rails. That will dissipate a decent amount of heat. Worth using the 7V5 SMPS as a source for the 5V rail?

hartytp, 2019-12-31:
@gkasprow I've completed my review.

Looks really nice!

gkasprow, 2019-12-31:
The Opamps consume max 14mA, which gives 28mA in total. With 7V losses on LDOs it gives us  2x 200mW.
SMPS would give +/-6V which would be converted to +/-5V by LDOs. SMPS losses would be roughly 300mW including quiescent power (1W SMPS have roughly 60% efficiency + 0.25W of idle power) + LDO losses of 2x28mW. So instead of 400mW we would have 360mW with increased EMI, cost and complexity. Not worth the effort IMO.

closed 2019-12-31 by gkasprow

hartytp, 2019-12-31:
Can you post an updated RC PDF at some point so I can sign off on it? Thanks!