sinara-hw/Pounder #51: consolidate rf amps

hartytp, 2019-12-31:
Can we just use MAR-8A+ everywhere instead of using ERA-2SM as well? This would save a BOM line.

gkasprow, 2019-12-31:
We can. Either we operate it with strong feedback or add an input attenuator.
MAR8 has very high gain or >30dB, we need 7dBm so the input signal would have to be attenuated by 20dB.

hartytp, 2019-12-31:
Well it’s not a big deal and we can keep things as they are if it’s a problem, just nice to keep the Bom short where we can

I think the attenuator is fine since we won’t be thermally limited with a nf of 23dB when using the dds pll. I’m any case the rf input has an attenuator that must be used with large signals so I don’t think the lo will limit.

Feedback fine by me if you prefer.

I’ll leave the choice (do nothing, change amp and add attenuator/fb) to you.

gkasprow, 2019-12-31:
I'd leave it as it is now. The MAR-8A+ has large gain variation vs frequency

hartytp, 2019-12-31:
@gkasprow if we're reducing the RF power anyway, would it make sense to use 2xERA-2SM? It's a slightly nicer amp and will do both jobs AFAICT.

Anyway, feel free not to bother, just throwing it out there...

gkasprow, 2020-01-01:
double ERA-2SM will significantly increase power consumption

closed 2020-01-01 by hartytp