sinara-hw/Pounder #52: LO power

hartytp, 2019-12-31:
Why 11dBm? Isn't that a little high for a level 7 mixer?

hartytp, 2019-12-31:
Isn't 10dBm a little high for the RF? What's the conversion loss when it's that saturated?

gkasprow, 2019-12-31:
True, I don't remember where these numbers came from.

gkasprow, 2019-12-31:
I fixed the LO part. I used 6 dB attenuator to have LO at 7dBm.
However, the DS does not say anything about RF power. IP3 is 15dBm so with 10dBm we are too close to it. Let's stick with 0dBm of output power and 5dBm of input power. I used such passive mixer even with power of -50dBm.
0dBm gives 632mV pp so we need 3x gain after the mixer.

dnadlinger, 2019-12-31:
> Why 11dBm? Isn't that a little high for a level 7 mixer?

That probably comes from what I had suggested as the solution optimised for close-in fibre noise cancellation performance. (If you use the mixer solely as a phase detector, you want to saturate the LO; the 11 dBm seemed like a good choice based on measurements I found by NIST/Rubiola/…). 7 dBm should also be fine for now.

hartytp, 2019-12-31:
@dnadlinger okay. Well if you'd prefer to revert then it's fine by me. Just let us know what LO/RF power you want.

dnadlinger, 2019-12-31:
7 dBm is fine.

hartytp, 2019-12-31:
I have no experience (either first-hand or from reading) of driving mixers outside their normal power recommendations, so that's what I'd tend to do -- although it doesn't make it correct for any given application and I don't feel strongly.

@gkasprow how about we go for +9dBm? As the data sheet shows, performance is slightly better with higher LO powers. This power is about right since it means that even with 1dB of deviation from nominal (e.g. amp gain variations) we'll still be comfortably above +7dBm, and below the +10dBm max they characterise the mixer at.

gkasprow, 2020-01-01:
Let's go for 9dBm then.

closed 2020-01-01 by gkasprow