sinara-hw/Pounder #56: loop filter

hartytp, 2019-12-31:
why do the DDS loop filter default population options not follow the eval board?

gkasprow, 2019-12-31:
well, it does
[image]
[image]

hartytp, 2019-12-31:
Iirc the eval design connects the cap to a vcc

gkasprow, 2019-12-31:
Here is [a schematic ](https://www.analog.com/media/en/technical-documentation/evaluation-documentation/57418637811849AD9959_pcb_0.pdf)of eval board. The drawing above was copied from that pdf file.

hartytp, 2019-12-31:
hmmm...I was looking here https://www.analog.com/media/en/technical-documentation/evaluation-documentation/ad9958-ad9959-schematic.pdf

[image]

gkasprow, 2019-12-31:
OK, let's connect it to CLK_VDD then.

hartytp, 2019-12-31:
Okay, so there are two inconsistent versions of the schematic linked to from https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/eval-ad9959.html#eb-documentation

The one I posted matches the pin description 
[image]

Also, IIRC it's more normal to decouple the CP output to Vcc than to GND so it makes more sense that way.

Maybe the schematic in the eval board data sheet is out of date?

hartytp, 2020-01-09:
If we're going to follow the eval board/pin description, shouldn't the resistors be 0R?

gkasprow, 2020-01-15:
OK, let's make it 0R
[image]

closed 2020-01-15 by gkasprow