sinara-hw/Pounder #63: run PI sim

hartytp, 2019-12-31:
(if only to check power margin on LDOs etc)

gkasprow, 2019-12-31:
Setting up the PI analysis takes a few hours of work so I do it only for complex circuits or critical rails. Here, we are copying power supply from already tested designs. I don't think it is necessary in this case.

hartytp, 2019-12-31:
Ok fair enough

Please can you 
- [ ] add a power budget to the schematic
- [ ] confirm that no rail exceeds its max current 

Then we can close

hartytp, 2020-01-09:
The power budget says that the 5V0A rail can draw up to 145mA, but the LT1761 is only a 100mA regulator...replace with a 3045 (thereby removing a BOM line)?

hartytp, 2020-01-09:
All other rails look fine (we're generally using 3A LDOs for rails drawing 100mA or so!).

hartytp, 2020-01-10:
NB if we add a new pre-amp we'll need to check the power budget on that rail...

gkasprow, 2020-01-15:
good catch. Fixed
[image]

closed 2020-01-15 by gkasprow

hartytp, 2020-01-15:
Did you mean to connect pins 1 and 2?

gkasprow, 2020-01-16:
I did it already.