sinara-hw/Pounder #66: Panel

gkasprow, 2019-12-31:
Is the panel annotation OK?

hartytp, 2019-12-31:
- [ ] I didn't see a hw revision number
- [ ] assuming people have ideas about what the LEDs will be used for, it might be nice to label them by function
- [ ] I'd probably use In/Out rather than RF/FB as I think that's clearer
- [ ] For the ADC just just use ADC0 and ADC1 rather than P/N?

Otherwise LGTM.

gkasprow, 2020-01-01:
the HW rev is on the very bottom

gkasprow, 2020-01-01:
What LED functions can we have for Pounder?
Maybe Lock status? Detector signal level indication,
LED0, LED2, LED 4 are green
LED1, LED3, LED5 are red

dtcallcock, 2020-01-02:
Given the Stabilizer LEDs are labeled L0-3, should the Pounder ones be labeled L4 onwards?  Or at least something that won't lead to ambiguities.

> Lock status

Wouldn't that be one of the Stabilizer LEDs as usual.

> Detector signal level indication

Sure, that could be useful.

gkasprow, 2020-01-05:
LEDs are labeled L4-L9. Let's leave them as they are. Panel sticker can be laser-cut and added easily later on.

closed 2020-01-05 by gkasprow