sinara-hw/Pounder #68: question: output power levels

hartytp, 2020-01-09:
@dtcallcock @dnadlinger Maybe I've forgotten our original discussion on this, but what was the decision about the modulator RF output power level? Did we want a pre-amp there to boost this up to ~10dBm (enough to drive a nice resonant EOM) or are we fine committing to using an external amp?

Maybe it's too late to change this now...but I was curious.

dnadlinger, 2020-01-10:
I don't mind either way, and haven't been following the discussion.

dtcallcock, 2020-01-10:
Yes, I thought we had decided to scrap the 'power' amp (LHA-13HLN) and use a low power amp that's already on the BOM.  -6dBm max output is a bit weedy.

The ERA-2SM will work nicely as it has +16 dB of gain so will let us run up to +10dBm.

Sorry I didn't catch this earlier.  Is it too late to add @gkasprow  Not the end of the world if it is.

hartytp, 2020-01-10:
:+1: other than that all happy to close this design round?

gkasprow, 2020-01-15:
For this reason, I kept it under the compile mask. I can add it easily. I added 0R bypass resistor so one can keep the amplifier DNP

closed 2020-01-15 by gkasprow

gkasprow, 2020-01-15:
[image]