sinara-hw/Pounder #69: rc2

hartytp, 2020-01-09:
Completed my review of rc2. Looks good!

#63 needs to be resolved. The other open issues are non-critical as far as I'm concerned...

dtcallcock, 2020-01-10:
Annotation at C82 is -16dBm.  Should be -26dBm.

dtcallcock, 2020-01-10:
Should there be some cool logos on the board?

Other than that, I also think it looks great.

dhslichter, 2020-01-10:
[image]

dhslichter, 2020-01-10:
Make some line art of this guy.  He'd be pleased to be on the board :)

gkasprow, 2020-01-15:
is that OK?
[image]

closed 2020-01-15 by gkasprow

hartytp, 2020-01-17:
@gkasprow did you really take @dhslichter's comment seriously???? Can we scrap that plz?

On a more important note, I've seen jumpers like this lead to GHz-frequency oscillations before due to added input-output coupling
[image]

gkasprow, 2020-01-17:
Did U see the physical layout?
I can use bigger footprint that woulf not change amplifier pads distance. We can also scrap it. I thought that you dont need this amp.

Pozdrawiam,
G.Kasprowicz

[image]<https://user-images.githubusercontent.com/21218399/72594811-4e9e4580-3900-11ea-8d69-c0e84f81558b.png>

hartytp, 2020-01-17:
> Did U see the physical layout?

No. I don't have altium right now and you didn't include the layout in the pdf. If you're happy with this then it's fine by me.

> I thought that you dont need this amp.

We will want this amp for pounder (not for phaser, because it's a different application in our lab).