sinara-hw/Pounder #71: Shielding Can

dtcallcock, 2020-01-10:
I'm not sure the current shielding cans are optimal.  The main problem with PDH locks in output->input crosstalk.  Therefore to be useful I think they should cover the input only.  This is also the part of the circuit most sensitive to pickup. 

If it's a big job to alter we can always make measurements on v1.0 to see if it's worth the effort.

gkasprow, 2020-01-15:
It would be a lot of work to cover only Rx part. Especially when I added the PWR amp.
What I can quickly do is to move the Rx parts to the bottom layer, especially Rx transformer and PWR stage that can generate EMI

dtcallcock, 2020-01-15:
To be clear, you mean move this stuff to the other side of the PCB?

[image]

gkasprow, 2020-01-15:
no, just PWR amp and balun. They are close to the input

gkasprow, 2020-01-15:
ICs don't generate much coupling

dtcallcock, 2020-01-15:
Sounds like a reasonable compromise.

closed 2020-01-15 by gkasprow

gkasprow, 2020-01-16:
I drafted the new release. I think it can go to production.

gkasprow, 2020-01-16:
How many pcs shall we produce?

dtcallcock, 2020-01-16:
> I drafted the new release. I think it can go to production.

Sounds good to me.

> How many pcs shall we produce?

I'd say 3 is a minimum.  @jordens will need one for gateware development, and then us and Oxford would want one each for testing (assuming @dnadlinger or @hartytp is interested in having a play).  Probably all 3 should go to @jordens to start with in case there is a bad board.  

Whether you should make a few more on top of that depends on the price breaks you get for runs of this size.

dtcallcock, 2020-01-16:
Related: @jordens do you have Stabilizer v1.1 hardware available or do you need me to ship you one?

jordens, 2020-01-16:
I'm not going to rework boards.

jordens, 2020-01-16:
@dtcallcock I have a stabilizer 1.1. Can you port the current firmware to it?

dtcallcock, 2020-01-16:
> I'm not going to rework boards.

Exactly.  If you have more than one board you can put a bad board aside and use a different one instead of reworking\debugging the bad one.

dtcallcock, 2020-01-16:
> Can you port the current firmware to it?

No, I don't have anyone with those skills in my group yet.  How much work is involved?

jordens, 2020-01-16:
In the past the prototype fabrication has been consistent. It makes more sense to hold two boards at the Fab for possible reworks.

dtcallcock, 2020-01-16:
> It makes more sense to hold two boards at the Fab for possible reworks.

Then let's do that.

jordens, 2020-01-21:
@dtcallcock see quartiq/stabilizer#66