sinara-hw/Pounder #76: SYNC_OUT to Stabilizer counter timer input

jordens, 2020-02-27:
* I don't see how SYNC_IN is useful
* We should enable SYNC_OUT to the CPU for accurate software-demodulation (required for quartiq/stabilizer#80)
* This needs a timer channel input with external clocking capability to be used as counter
* Counting 125 MHz may or may not be doable with the stm32h7 (but should be checked) physically.
* From what I can tell it's not possible with the HRTIM. It would need to be a general purpose or advanced timer
* Alternative is (a) a prescaler or (b) using one of the the reference clock outputs from Pounder to a timer channel counter input on Stabilizer

c.f. #55

gkasprow, 2020-08-17:
@jordens I plan a new release of Stabilizer. Which solution would be the most promising? An LVCMOS divider /8 would be fine?
Of course, I would have to change in Stabilizer design the IO pin assigned to this particular GPIO header pin. I can add such circuit to existing boards - I have one Stabilizer and one Pounder.
What do you mean by prescaler phase measurement and compensation? D you mean IC propagation delay? Why would we care about it since the STM32 input is not compensated.

jordens, 2020-08-24:
We use PA0 for DDS_RESET on Pounder v1.0. After we checked the max input frequency on PA0, I'd suggest connecting DDS_RESET to J10:11 (PG6) and SYNC_CLK (if needed through a /8 prescaler) to J10:18 (PA0). And on Stabilizer the PA0 wire should be cleaned up to and there should probably be some termination (series or AC parallel?).

gkasprow, 2020-09-01:
It's easier for me to add a prescaller and bypass option than to check max STM frequency, especially if it is undocumented and may not work with other production batches.

gkasprow, 2020-10-10:
@jordens I cleaned the PA0, the next action is to modify the pounder, right?

jordens, 2020-10-11:
Yes

jordens, 2020-10-12:
I checked the counter input. No external prescaler needed. Just

* connect DDS_RESET to J10:11 (PG6)
* connect SYNC_CLK to J10:18 (PA0) make sure that this 125 MHz signal gets to the CPU with correct level and decent termination/matching

ryan-summers, 2020-12-09:
The changes @jordens proposed above were manually completed by me on TS 38/20 0002. In addition to the changes mentioned above, I also removed R70 (to disconnect SYNC_IN from PA0).

I have run firmware and verified that the clock can be properly driven using this hardware configuration and this is sufficient for timestamping ADC samples.

closed 2021-01-23 by gkasprow