sinara-hw/Pounder #79: IF amp oscillating

jordens, 2020-05-14:
@gkasprow you [mentioned](https://github.com/sinara-hw/Pounder/issues/19#issuecomment-569631706) the THS4631 is weird.
It oscillates when the IF gets to -0.4 V (at the output, G=3). A couple pF negative feedback fix it.

[image]

gkasprow, 2020-07-16:
The THS4631 input is weird. You cannot make classical integrator out of it for example due to strange non-lineairty.

gkasprow, 2021-01-23:
I added 4.7pF mounted by default

gkasprow, 2021-01-23:
There is new fast JFET opamp which behaaves as expected - it's OPA818

jordens, 2021-01-23:
THS4631DGN is still in the schematic v1.1rc1. don't you want to use the opa818?

gkasprow, 2021-01-23:
It has a completely incompatible footprint. I can add it as an assembly option and populate it by default.

jordens, 2021-01-24:
Maybe there are footprint compatible alternatives? The oscillation is really a functional problem. Or can we change the feedback to reliably prevent it?

gkasprow, 2021-01-24:
OPA657, AD8065

jordens, 2021-01-24:
AD8065 for lower power. I don't need offset stability or robustness against RF rectification. Can you add 100 or 1k series protection to ADC_[01]_P? The 50R termination would otherwise exceed abs max rating.

jordens, 2021-01-24:
Let's go for the AD8065.

gkasprow, 2021-02-03:
ok, done
added 50R resistors and replaced an opamp

closed 2021-02-03 by gkasprow