sinara-hw/Pounder #81: Missing level translation between STM32 and attenuators

dtcallcock, 2020-06-11:
See here: https://github.com/quartiq/stabilizer/issues/103

hartytp, 2020-06-12:
Isn't this the same as Urukul?

Interfacing between 5V and 3V3 logic directly isn't an issue so long as all logic thresholds are met and no absolute max rating is exceeded. @gkasprow or @jordens probably remember better than me, but IIRC this was all considered at design time; the logic levels are all fine and the 470Ohm resistors (in combination with the input clamping diodes) were chosen to be large enough to protect the CPLD/microprocessor inputs from the 5V outputs. So, this should all work fine without need for additional translators.

gkasprow, 2020-06-13:
Attenuators accept 3V3 logic, H state requires 3V or more. Exactly, 470R is used to avoid clamping

dnadlinger, 2020-06-13:
Can be closed, then?

jordens, 2020-06-15:
Are the logic levels fine? Certainly not on paper.
The attenuators want `V_ih = 3 V`, STM32H7 gives `V_oh = V_dd - 0.4 V = 2.9 V`. Same for the CPLD.

hartytp, 2020-06-15:
@jordens ok, thanks for elaborating. That does sound a bit marginal then.

gkasprow, 2020-06-16:
These Vdd-0.4 are usually specified under resistive load. We don't have such.

gkasprow, 2020-06-16:
We can add pullup to fix the potential issues

gkasprow, 2021-01-23:
added 10k pullups on MOSI and SCK

closed 2021-01-23 by gkasprow