sinara-hw/Pounder #84: v1.1

jbqubit, 2020-12-09:
Talking to @dtcallcock about v1.1. @gkasprow discussed the following change.  https://github.com/sinara-hw/Pounder/issues/76#issuecomment-675129572

Are there are additional tests that need to be done on v1.0 before v1.1?

dtcallcock, 2020-12-09:
It would be good to measure the noise floor with output connected to input and phase adjusted to zero mixer output.   I'd do this both with an external instrument (via Stabilizer's analog in SMA) and via the Stabilizer's ADC.  I'm afraid I don't have boards or people to do this myself in the near term.

https://doi.org/10.1109/FCS.2011.5977868 gives an idea of how you might do this and where you'd expect to get to if the mixer was limiting.  The DDS is probably the limit here - see http://rubiola.org/pdf-slides/2012C-IFCS-DDS.pdf.  Hopefully the shared clock circuitry, power supplies, and thermal environment make some of the noise sources highlighted by Rubiola common-mode out, but AFAIK nobody in the literature has tested this.

jbqubit, 2020-12-09:
Who has copies of Pounder [v1.0](https://github.com/sinara-hw/Pounder/releases/tag/v1.0rc3)? I've checked with Technosystem and they have no v1.0 available.  Seems natural to apply round up testing and bug fixes for new round of v1.1 PCBs.

dtcallcock, 2020-12-11:
The phase detector sensitivity would be part of this measurement and should be documented (as a function of frequency).

This issue should probably be revisited at that point too: https://github.com/sinara-hw/Pounder/issues/28

gkasprow, 2021-01-22:
I discovered that have one copy of Pounder as well as Stabilizer. I also have  at home the VNA, SA and MW generator (square wave only) so can do some tests, but will need some help with software setup.

closed 2023-09-21 by maciejprzybysz

maciejprzybysz, 2023-10-23:
I closed this issue as it was about v1.1 and not updated over 2y.

In case this is still worth to do I made a separate issue for measurement task: https://github.com/sinara-hw/Pounder/issues/109