sinara-hw/Pounder #85: measure noise floor

jbqubit, 2020-12-09:
cf https://github.com/sinara-hw/Pounder/issues/84#issuecomment-741345552

> It would be good to measure the noise floor with output connected to input and phase adjusted to zero mixer output.   I'd do this both with an external instrument (via Stabilizer's analog in SMA) and via the Stabilizer's ADC.  I'm afraid I don't have boards or people to do this myself in the near term.

> https://doi.org/10.1109/FCS.2011.5977868 gives an idea of how you might do this and where you'd expect to get to if the mixer was limiting.  The DDS is probably the limit here - see http://rubiola.org/pdf-slides/2012C-IFCS-DDS.pdf.  Hopefully the shared clock circuitry, power supplies, and thermal environment make some of the noise sources highlighted by Rubiola common-mode out, but AFAIK nobody in the literature has tested this.

kl2025, 2021-08-17:
Hello, as an intern in mlabs, I am following the above paper to do a quick test to measure the noise floor of the Pounder's mixer_ch_0 output, which is connected to the Stabilizer ADC_0. 

My hardware test setup: 
I am using Stabilizer v1.2 and Pounder v1.1. They are stacked together and Pounder OUT_0 is connected to IN_0 with a SMA cable. 50 ohm terminations are switched on for both stabilizer ADC channels. 

[image]

My software test setup:
I am using Quartiq Stabilizer firmware and udp datastream to log the Stabilizer ADC data. So the Stabilizer ADC sampling rate is around 781.25 kHz or 1 data per 1.28 us. 
Pounder OUT_0 is basically DDS_CH_0_OUT plus an attenuator. It is the RF source going to the mixer (through a SMA cable to Pounder IN_0) and it is set to output sine waves with frequency = 5 MHz, phase = around pi/2 (adjust until no dc in the mixer output) , amplitude = -16 dBm and output attenuation = 0 dB. 
Pounder IN_0 has two signal paths: RF attenuator and internal LO source from DDS_CH1_OUT. RF attenuation is set at 10 dB so that the RF will become the suggested value of -26 dBm before the +31 dB amp. The LO is set to have frequency = 5 MHz, phase = 0, amplitude = -3 dBm to match the suggested value. 

Both LO and RF signals meet at the mixer and the mixer IF output is logged from the Stabilizer ADC_0 data streaming. 
Here are Stabilizer ADC_0 plot with ADC_0 gain set to 10 and its FFT spectrum using np.fft,rfft in python. 

[image]

we guess the FFT peak at around 0.055 kHz is the 50 Hz mains hum. 

I also connect stabilizer ADC_0 to a SIGLENT SDS2074X Plus oscilloscope with a SMA coaxial cable to look at the mixer output.
The signal looks like this :  

[image]

A quick FFT of the signal in the oscilloscope is put here as a don't-be-serious reference. With limited oscilloscope bandwidth, I am not sure about the accuracy with this internal FFT function . So better check this again if someone has a proper spectrum analyzer. 

[image]

It's my first time to do such noise measurement, feel free to let me know if there is anything I should improve, thank you.

gkasprow, 2022-07-29:
Measuring anything with scope requires caution. First, turn off the target's power supply but keep all the wires and loops the same. At least GND of the input power. MEasure with scope. Then turn on the power. MEasure again. You will often see very little difference due to the scope cable and ground loop pickup.