sinara-hw/Pounder #86: v1.1 changelog

gkasprow, 2021-01-23:
- aligned SMA connectors #80 
- added 22uF at the LDO outputs #77 
- moved PCB edge #75 
- fixed PN in 47k resistor #83 
- R134, R166 changed to 0402 to lower the BOM count

jordens, 2021-01-23:
Great. Could you spin the RC production files so we can give it a quick check?

gkasprow, 2021-01-23:
[done](https://github.com/sinara-hw/Pounder/releases/tag/v1.1rc1)

jordens, 2021-01-24:
I had a look at the changes. Apart from the opamp change #79 I'm ok with v1.2 and would like to proceed with getting it manufactured. As a note I still think that:

* The baluns should all become TC1-1T (replace 7 ADT1-1WT)
* The use of pricey "special RF" capacitors for DC-blocks at 1-200 MHz is not supported by data, evidence, or experience with the other dozen designs that don't have and don't seem to require them. When I look at standard MLCC for DC-blocking in 1-200 MHz, they are all just fine.
* Putting RF output and RF input signal paths into the same RF shield box makes the purpose and benefit of the boxes unclear to me: is it for emission or immission or crosstalk?
* The RF output amplifiers can go away.

But since this is all still experimental, I'm not going to fight these.

jordens, 2021-02-03:
@gkasprow AFAICT there is only the opamp change (#79) pending for v1.2. Can we cut the release and send the production files?

gkasprow, 2021-02-03:
> The baluns should all become TC1-1T (replace 7 ADT1-1WT)

Just to simplify the BOM?

>    The use of pricey "special RF" capacitors for DC-blocks at 1-200 MHz is not supported by data, evidence, or experience with the other dozen designs that don't have and don't seem to require them. When I look at standard MLCC for DC-blocking in 1-200 MHz, they are all just fine.

I was curious about real benefits. We can always use Hi-Q caps instead.

>    Putting RF output and RF input signal paths into the same RF shield box makes the purpose and benefit of the boxes unclear to me: is it for emission or immission or crosstalk?

Only output path attenuator was placed under the shield, the PA sits on the other PCB side. If we discover crosstalk, I will place the attenuator on the bottom PCB side

>    The RF output amplifiers can go away.

Why?

jordens, 2021-02-03:
> Just to simplify the BOM?

BOM simplification, availability (the ADT1-1WT are quite a bit more exotic and hard to replace with anything footprint compatible, while the TC1-1T style is ubiquitous), and board space.

>  If we discover crosstalk

Right. We should measure crosstalk.

> The RF output amplifiers can go away.

They are typically not needed. Like for Urukul and Mirny, the usual amplifiers (e.g. Booster) would take 0 dBm (-5 dBm might be a little low however). And power usage: getting rid of them would reduce the power consumption by 0.7 W.

gkasprow, 2021-02-03:
The amplifier can be bypassed by the assembly option. I will make the bypass default configuration.

dtcallcock, 2021-02-03:
> They are typically not needed. 

Actually I find the -5 to 10dBm range is often what you need to add few percent sidebands with a modulator device/laser input, and having to add an external amp to achieve these very modest signal levels would be annoying.  But as long as it's retained as an option then that's fine.  

Noting on the schematic/wiki that an [ERA 50SM+](https://www.minicircuits.com/pdfs/ERA-50SM+.pdf) can be populated here instead would also be nice.  This extends the output to an even more useful +13dBm.  Sure it's another 280mW of power consumption but it's just a suggestion so nobody needs to compromise.

gkasprow, 2021-02-03:
[v1.1 release is ready](https://github.com/sinara-hw/Pounder/releases/tag/v1.1rc2)

gkasprow, 2021-02-03:
@kaolpr @akaminska

dtcallcock, 2021-02-03:
[image]
Super minor, but can you make R34 the omitted component, not L15?  Any reasonable lab will have a kit with suitable resistors but the right inductor would probably have to be ordered.

gkasprow, 2021-02-03:
the inductor will make a stub

dtcallcock, 2021-02-03:
true dat

gkasprow, 2021-02-03:
One would have to order the amplifier anyway :)

dtcallcock, 2021-02-03:
Good point. Perhaps add the part number as a schematic annotation so one can order it without having to find an Altium licence though?

dnadlinger, 2021-02-04:
Shouldn't we be able to just flip them around? A 82 Ohm resistor should be low-enough Q at our frequencies? Just asking out of curiosity.

gkasprow, 2021-03-04:
yes, we can swap them. The resistor has smaller pads than the inductor so capacitance should be not an issue
[image]

closed 2023-09-21 by maciejprzybysz