sinara-hw/Pounder #89: trace impedance and separation

jordens, 2021-02-12:
The 50 Ohm signal traces seem to be really close to other traces, other pads, copper fill elements. Are the rules ok?

[image]
[image]
[image]

gkasprow, 2021-02-14:
it looks so.

jordens, 2021-02-14:
Yes. Coupling to the GND plane below may dominate making this OK.
Let's leave it open as one item for characterization: crosstalk. If that turns out to be problematic, we can look at swapping of the in/out DDS channels (much more separated routing without layer crossing jumpers) and separated shielding cans.

[image]

gkasprow, 2021-03-04:
I fixed the 3 first issues

gkasprow, 2022-07-29:
I improved trace routing - added GND copper shielding
[image]

jordens, 2022-07-29:
Nice. Thanks.

closed 2022-07-29 by gkasprow