sinara-hw/Pounder #93: Pounder may seat properly with pins incorrectly aligned

ryan-summers, 2021-04-08:
When mounting Pounder onto stabilizer, it's possible to inadvertently connect Pounder incorrectly onto the Stabilizer headers. Specifically, Pounder will properly seat 1 pin too high or low on the board.

When this occurs, it can be difficult to visibly see the alignment issue, but I have observed that my power supply begins to encounter over-current conditions on boot (presumably due to some short-circuit condition). We should likely select components such that incorrectly seating the device is not possible.

[image]
[image]
[image]

jordens, 2021-04-08:
Don't the connectors interfere? After all those are tub-style and I don't see how they could mate with an offset.

ryan-summers, 2021-04-08:
There's enough clearance such that it still fits. I don't notice much difference in the force required either.

jordens, 2021-04-08:
Ok. Then let's look for keyed female connectors.

gkasprow, 2021-04-08:
Are you able to mount spacers? I don't think so. They are included in the BOM, so should be mounted.

gkasprow, 2021-04-08:
I tried to find suitable connectors with the right height and it was tough. Maybe would be good to glue in 2mm pieces of some plastics into the Stabilizer connector edges.

jordens, 2021-04-10:
The spacers were not supplied.

jordens, 2021-05-31:
Maybe we could just use an oversized Samtec ESW-1xx-23-L-D to resolve this. Since the current size fits fine at +-1 row offset, we should just use that part with two more (unused) rows so it can't be inserted at an offset.

gkasprow, 2022-07-31:
I added holes that enable assembly of oversized connections