sinara-hw/Pounder #94: Temperature measurement with power detector

jordens, 2021-05-31:
The power detectors are reasonably close to the RF sections. I'd like to use them for temperature monitoring in a precision measurement. Let's connect their temperature PTAT output to the unused CPU ADC inputs.

* Appropriate voltage divider 5V to 3V.
* We can use IN2_N in single ended mode.
* IC10A TEMP to JADC1_IN2_N
* IC10B TEMP to JADC2_IN2_N

jordens, 2022-07-28:
You are right. Divide to 2.048 V

gkasprow, 2022-07-28:
2k and 3k are a bit low, but we already have them in BOM
[image]

jordens, 2022-07-28:
@gkasprow I had another look.
For 125 C it would be 1.9 V max. And we have protection diodes there. Just go with 10k series. No divider needed.

[image]

gkasprow, 2022-07-28:
OK

closed 2022-07-28 by gkasprow