sinara-hw/Pounder #95: Power measurement redesign

jordens, 2021-05-31:
The power measurement circuit doesn't seem to work well. I'm mostly measurring ~1V output on the detector almost independent of how much power is at the input.

* The impedance isn't matched. The tap has 525 R output impedance, the detector has 50 R. One would think that's fine given the short (~20 mm) trace but it looks weird. Let's make it a proper discrete 1:10 splitter
* Let's place the detector tap after the preamps. No need to loose more precious power at the input. The working power range is also badly matched to what's typically available before the amplifier including the tap loss.
* The detectors are pretty hungry (350 mW each). Can we use their power down interface with GPA7 and GPA6 of the gpio extender?

jordens, 2021-06-04:
* VTGT: Due to the input bias current and input impedance the voltage is 1.35 V (instead of 1.4 V), close to the minimum of 1.3 V. The eval board has a lower impedance divider (1k4 || 845R) than ours (5k6 || 3k6). Let's take the voltage divider from the eval board.

jordens, 2021-06-04:
* The VOUT is 0-5 V (maybe 0-3 in practice) output. The JADC[12]_IN2_P are 0-2 V. Add a 2/5 voltage divider.

gkasprow, 2022-07-28:
the TCM2 pin requires nearly 5V to disable the chip.
I used more common resistors to get 1.4V from 2.3V

[image]

gkasprow, 2022-07-28:
@jordens I assume you are talking about such splitter
[image]

gkasprow, 2022-07-28:
it has 20dB tap and 1.7dB insertion loss

jordens, 2022-07-28:
Exactly. All R close together.

closed 2022-07-29 by gkasprow

MatBier, 2023-01-04:
@jordens @gkasprow 
I encountered a problem with power measurement when testing Pounder 1v2.
I've tested 3 boards and on all of them output voltage from power detector is stuck on max voltage (around 4.3V - 4.8V depending on the board). Output is not affected by variable input power and attenuation level. Problem seems to look like one from May 31, 2021 comment, but I also checked power measurements on one Pounder 1v0 board and I was able to detect it normally.

I changed resistor values in power splitter on one of 1v2 PCBs but it doesn't seem to help.

jordens, 2023-01-04:
I haven't tested the new power detector design.

jordens, 2023-01-04:
Maybe it's in power down mode?

MatBier, 2023-01-05:
It is most definitely in operational mode, when in power down output is 0V.

gkasprow, 2023-01-05:
The difference between these two designs is the way how RF signal is sourced.
New revision gets the signal after IC2 amplifier so it means it is nearly 30dB stronger. Instead of -26dBm -  20dB = 46dBm now we get 5dBm-20dB = -15dBm. That should be still within the range of the detector

gkasprow, 2023-01-05:
we also changed the VTGT value from 1.35V to 1.4V. Please check if pin 12 is 1.4V

gkasprow, 2023-01-05:
Please also check voltage on pins 1 and 16

MatBier, 2023-01-10:
Turns out voltage on pin 12 is 0.88V. There is a mistake on the schematic - R16 and R63 values should be switched. With proper resistors power meter works properly.