sinara-hw/Pounder #99: IF amp filter and gain

jordens, 2021-05-31:
* This should read "0 dBm ..."
[image]

* At the sum mixer product (of >20 MHz) it's OK and beneficial to use very low impedance. That makes the mixer happy and also helps with suppressing recification in the opamp. Let's make a 10 MHz Chebyshev pi-filter:
  * DNP R22A,B
  * R62A,B = 49R4
  * R63A,B = 0R
  * C27A,B, C28A,B around 560 pF
  * L8A,B around 1 µH

* IMO the IF amp gain should be increased from 3 to 10 for all use cases:
  * Only the double-saturated use case of the mixer will get anywhere near 0 dBm output. All other use cases will need to stay significantly below (like -6 dBm).
  * We have the in-amp on Stabilizer that can cope fine with +-3V.

jordens, 2021-06-01:
We could also consider a proper anti-aliasing lowpass here.

[image]

jordens, 2021-06-05:
And I would like to consider a better noise match to the 50R mixer IF impedance. The FET opas are cool for high impedance sources but that's not what we have here. DC offset isn't a big problem since it's typically dwarfed by the mixer DC offset (~0.5 mV). Rectification to me seems to be the same effect as what the mixer itself would be limited by.

gkasprow, 2022-07-28:
@jordens where would you put such LC anti-alias filter? Shall I add more LC sections before the opamp? I can add another placeholder for LC section.

gkasprow, 2022-07-28:
In terms of an opamp, what about [ADA4896/7  ](https://www.analog.com/media/en/technical-documentation/data-sheets/ADA4896-2_4897-1_4897-2.PDF)

jordens, 2022-07-28:
Let's go for this filter:

[image]

* Plase that filter between mixer IF and opamp. The only resistor should be a RL = 50 R termination but after the filter. Keep C60 at the opamp input. R21=0R. Remove R9 traces (potential for annoying positive feedback). Make the opamp with gain=20 (to also lift it over the INA noise which is best case 18 nV/rtHz), cap its gain to 1 MHz.

[image]

* Make sure the mixer sum products (5-500 MHz) see low impedance (< 50R, can be capacitive). Shunt-first Chebychev helps here, but let's put a 33 pF in parallel to that first big shunt C.
* [ADA4896/7](https://www.analog.com/media/en/technical-documentation/data-sheets/ADA4896-2_4897-1_4897-2.PDF) looks good to me in gain=20

gkasprow, 2022-07-29:
I got rid of the notch filter and other options to make space
[image]

jordens, 2022-07-29:
Perfect

closed 2022-07-29 by gkasprow