sinara-hw/Pounder #100: Square up LO

jordens, 2021-06-10:
To reduce the mixer temperature coefficients as well as LO AM-PM sensitivity, it would be very useful to increase the slew rate and square it up.

[image]

This could be done in differential signalling. In that case we would want a differential AA filter or two matched filters and a clock buffer. If possible, that clock buffer would replace the amplifier IC13A/B.
Maybe we can even go differentially (differential clock buffer) into the mixer LO at significant benefit (LO isolation and amplifier swing). We should just double-check that the GND pin opposing LO is actually the other end of the LO balun winding).

[image]
[image]
[image]

gkasprow, 2022-07-28:
MAybe use sth like [LTC6957](https://www.analog.com/media/en/technical-documentation/data-sheets/6957fb.pdf)?
LTC6957-4 has CMOS Logic, Complementary Outputs

gkasprow, 2022-07-28:
@jordens, what cut-off frequency would you like to have?

gkasprow, 2022-07-28:
what's wrong with the LTCC filters we use right now? We can use more of them to get rid of aliasing.

jordens, 2022-07-28:
The LTCC filters are fine but single ended. With a balanced filter, differential (in and out) clock buffer we'd be all differential and save the balun as well.
But honestly idea is a bit more experimental and risky than I'd dare put into the next iteration without prototyping it first. Let's leave it out and open for now. Maybe somebody wants to pursue this as an experiment. I'd certainly be interested in seeing the result.

gkasprow, 2022-07-29:
I thought you want to put the filter before the buffer to make perfect sine and then convert it to perfect square wave.

jordens, 2022-07-29:
Yes. Filter before buffer.
My goal with the balanced filters would have been to have then entire chain (DDS, filter, buffer, Mixer LO) balanced and never convert to/from unbalanced. That's what they do

gkasprow, 2022-07-29:
OK. it's better to design a simple PCB with components and connect only IN, OUT and supply
I't put there entire chain; filter, clock buffer, mixer, and output filter. I can also shape it in such way that you can solder it on top of Phaser board without using long wires. - just remove old components from PCB and fit everything under the shield.
the only signal that would need to be delivered is twisted pair from DDS

jordens, 2022-07-29:
If you can find the space on the board (for a balanced filter + buffer as a bypass to the chain of dds output balun + LTCC filter + pad + RF amplifier) then add them. Otherwise I'd say leave it as is.

[LMK1D2102RGTR](https://www.ti.com/lit/ds/symlink/lmk1d2102.pdf?ts=1659002864882&ref_url=https%253A%252F%252Fwww.google.com%252F) is also an as buffer.

gkasprow, 2022-07-29:
I have some place on the bottom so can try to install it there.

gkasprow, 2022-07-29:
I'm not sure if LVDS can drive 10dBm (1.2V peak) into a 100Ohm load.
It usually has only 4mA of current.
LVPECL is a better option and would give much better rise time than CMOS.
CMOS output is too strong and we would have to attenuate it.

jordens, 2022-07-29:
The mixer LO is 50 Ohm, also if we use it differential.

gkasprow, 2022-07-29:
right, that's even worse in case of LVDS

gkasprow, 2022-07-29:
What cutoff frequency do you want for the differential path? 160MHz?

gkasprow, 2022-07-29:
@jordens and what order of the filter do you need?

gkasprow, 2022-07-29:
@jordens what about such circuit?
[image]

gkasprow, 2022-07-29:
and filter characteristics
[image]

gkasprow, 2022-07-29:
the driver resistance is very low (open emitter). Maybe we should add two 24Ohm resistors to make it a 50Ohm differential...
The driver provides 1.6Vpp, but with matching that would be halved.

jordens, 2022-08-01:
The LO port is a very non-linear one anyway (it's two series pairs of diodes): VSWR against 50 Ohm 7 dBm is 2.5, and 50 Ohm 10 dBm is 3.5. As long as you saturate them well and the buffer chip is happy with the load, and there is no long lines between buffer and mixer, we should be fine without any termination (R100, 101, 102). Maybe series resistors to limit the current would be better.

As filter, with 250 MHz Nyquist, I'd go for something like this (balanced, i.e. doubled up; but component values don't matter now since it's a variant anyway).
[image]

gkasprow, 2022-08-01:
We need R100, 101, and 102 because they act as pulldown. And these pulldown resistors limit the current. The LVPECL is an open emitter output stage.
I marked these filter components as DNP and placed the values you suggested.

closed 2022-08-01 by gkasprow

jordens, 2022-08-01:
Ack. I got it now.
Looking at the VSWR the LO port of the mixer is more like an "average" 15 - 20 Ohm impedance for a sine wave at 10-7 dBm. As would be expected from 2 diodes. With that in mind R100/R102 could actually be large (e.g. 100 Ohm) to maximize power into LO and R101 to something lower like 0R to adjust the working point.
All up for tuning.

gkasprow, 2022-08-01:
When you look at the datasheet, the output voltage amplitude does not depend much on the load resistor. You only change the offset. But yes, a higher value will dissipate less power.

gkasprow, 2022-08-01:
I mean this plot
[image]

jordens, 2022-08-01:
Yep.